wpilibsuite/allwpilib #5264: PWM shows value in Sim Gui, but not NT.

blu28, 2023-04-12:
**Describe the bug**
I have a PWM that I set the value while running the Desktop simulator. The simulator correctly detects that a PWM is allocated and correctly displays the value. However, when I put the PWM on the Smartdashboard with Smartdashboard.putData(pwm), the network table entry never updates, it always shows zero. 

**To Reproduce**
In a Robot.java file instantiate a PWM object on any channel, then set it to be on the smartdashboard with a putData call. Then in robotPeriodic set the value with pwm.setSpeed(somevalue). The value will correctly show on the simulator GUI, but the network table entry will remain zero.

**Expected behavior**
I expect the network table entry to reflect the value that has been set.

**Desktop (please complete the following information):**
 - WPILib Version: 2023.4.2
 - OS: Windows 10
 - Java version 17.0.5
 - C++ version NA

**Additional context**
If I put a breakpoint in getRaw in PWM, I can see that the lambda in the Senadable Registry is getting called. The value being returned by PWNJNI.getPWMRaw(m_handle) is always zero.

sciencewhiz, 2023-04-15:
I suspect this has a similar root cause as #3606

blu28, 2023-04-15:
I have been coming to that conclusion as well. I think that there should be a basic tenet, that by default everything run in the simulation environment or in a unit test environment should have behavior as close as possible as the actual hardware. In particular people who use the simgui are going to expect that.  This is especially important since people developing software are often going to use the simgui to help them understand how the hardware actually behaves before they are able to get the actual hardware. 

I find it frustrating in that while the WPI code is open source and I can trace through the behavior of the software, as I drill down I always meet up with a JNI call into NI code and it just becomes a black box. I know that somehow these calls often lead to the device simulation code, but I have no idea how. So, when the behavior of these devices is exposed in the simgui and unit tests, I have to expect that they at least behave as the actual hardware would.

PeterJohnson, 2023-04-15:
The JNI calls lead to the HAL.  There are two versions of the HAL (both open source in the repo), one that’s used on the Rio which calls into the NI libraries, and one for simulation.  So while there is a black box in terms of what the NI libraries do, the simulation side is completely open source (there are no NI libraries involved in simulation).

ThadHouse, 2023-04-15:
Back in the cRIO days, the JNI calls were direct into the FPGA, but that hasn't been the case for the entirety of the roboRIO era. If you ever want to find a JNI call, the format is always Java_package_with_underscores_not_dots_ClassName_functionName.

So that getPWMRaw call is named Java_edu_wpi_first_hal_PWMJNI_getPWMRaw located at https://github.com/wpilibsuite/allwpilib/blob/main/hal/src/main/native/cpp/jni/PWMJNI.cpp#L195

Which you can then trace to the files Peter posted. With the athena folder being the roboRIO implementation, and the sim folder being the Sim implementation.

https://github.com/wpilibsuite/allwpilib/blob/main/hal/src/main/native/sim/PWM.cpp#L239

sciencewhiz, 2023-06-19:
#5298 attempts to demystify the JNI a little by adding javadocs to the JNI, and also pointing to the HAL function called.

PR #5277 (2023-07-10): Unify PWM simulation Speed, Position, and Raw
Setting one will set the others, like it does in real hardware 
Add tests for boundary conditions and conversions
Update PWM sendable implementation to include all forms 
Fixes #5264
Fixes #3606
commit f8e74e2f (2023-07-10): [hal] Unify PWM simulation Speed, Position, and Raw (#5277)

Setting one will set the others, like it does in real hardware.
Add tests for boundary conditions and conversions.
Update PWM sendable implementation to include all forms.
Fixes #5264
Fixes #3606
closed 2023-07-10 by PeterJohnson