wpilibsuite/allwpilib #5276: PS4 Controller mappings might be wrong

TheTripleV, 2023-04-19:


srimanachanta, 2023-04-20:
Can you elaborate on what might be wrong?

srimanachanta, 2023-04-22:
I am unable to reproduce the issue on MacOS with simulation. Axes seem to be correct in code when compared to the sim gui

TheTripleV, 2023-04-24:
I think this has to be verified with NI DS

TheTripleV, 2023-06-17:
Would anyone with a (PS4 Controller + NI DS) combo be able to try this out and record
1. each physical button press / joystick axis movement
2. movement shown in the NI DS joystick menu

ncgears, 2023-12-23:
Validated buttons with NI DS and PS4 Controller (USB)
Axis: 
LeftY = 1
LeftX = 0
RightY = 5
RightX = 2
LeftTrigger = 3 (also Button 7)
RightTrigger = 4 (also Button 8)

Button:
Square = 1
Cross = 2
Circle = 3
Triangle = 4
LeftBumber = 5
RightBumper = 6
LeftTrigger = 7
RightTrigger = 8
Share = 9
Options = 10
LeftStick = 11
RightStick = 12
Playstation = 13
Touchpad = 14

Starlight220, 2023-12-23:
@ncgears does this match the currently implemented indexes?

ncgears, 2023-12-23:
This appears to match in PS4Controiller.java and PS4Controller.cpp.  Furthermore, I looked at all the getter functions and they all look correct for which axis values they are returning. 
This was tested with a genuine PS4 dual-shock controller. I also had a 3rd party PS4 controller, Etpark Wired Controller for PS4, sold by REV for the FTC teams, REV-39-1865. This was detected in NI DS as "XBOX 360 For Windows".  The mappings on this controller are as follows:

Axis:
LeftY = 1
LeftX = 0
RightY = 5
RightX = 4  (note that this is different and could explain the issue!)
LeftTrigger = 2 (no button)
RightTrigger = 3 (no button)

Button:
Square = 3
Cross = 1
Circle = 2
Triangle = 4
LeftBumber = 5
RightBumper = 6
Share = 7
Options = 8
LeftStick = 9
RightStick = 10
Home(PS) = Not detected in NI DS (This is XBOX Home though)
Touchpad = Not detected in NI DS
Rumble works as expected

Starlight220, 2023-12-23:
Hmm
If it's detected as an Xbox, maybe the indexes match that?

ncgears, 2023-12-23:
It does not match the XBOX mappings. I think the logical step here is to check with @TheTripleV to see if we can confirm if this may be the case (genuine dual-shock vs. 3rd party look-alike).  This appears to be a knock-off controller that doesn't appear to implement the axis true to the original.

ncgears, 2023-12-23:
If we can positively confirm this issue, at minimum, we might be able to get Rev to put a *NOTE* on the page for the product indicating that it does not match the PS4 button IDs.

PeterJohnson, 2023-12-23:
We could add a note to the class documentation as well.

ThadHouse, 2023-12-23:
We should mark all controller classes that the mappings are only valid for official first party controllers. 3rd party controllers are not guaranteed to have the same mapping.

TheTripleV, 2023-12-30:
Should also go in known issues imo, since all xbox style controllers (MS, thirs-party) seem to work the same and this only affects third-party PS controllers.

ThadHouse, 2023-12-30:
I think just documenting that only first party controllers are supported by the class is fine. There isn't any guarantee that controllers other then the exact ones listed have the same mapping. Now yes microsoft does a better job, but thats not guaranteed.

PR #6112 (2023-12-30): Document only first party controllers are guaranteed to have correct mapping
Closes #5276
closed 2023-12-30 by PeterJohnson