wpilibsuite/allwpilib #5280: wpilog files fill up the rio and don't delete

TheTripleV, 2023-04-20:
Docs say "On startup, all existing log files where a DS has not been connected will be deleted. If there is less than 50 MB of free space on the target storage, FRC_ log files are deleted (oldest to newest) until there is 50 MB free OR there are 10 files remaining."

Team reports that the rio is full of log files and errors and they had to manually delete them.

me - happened to me too once

PeterJohnson, 2023-04-21:
Could be a bug in our detection.  The code is not that complicated. https://github.com/wpilibsuite/allwpilib/blob/main/wpilibj/src/main/java/edu/wpi/first/wpilibj/DataLogManager.java#L238-L270

TheTripleV, 2023-04-24:
We could delete more aggressively and only skip over competition logs.

Additionally, it could compress old logs.

Autumn-Ou, 2023-04-25:
> https://github.com/wpilibsuite/allwpilib/blob/main/wpilibj/src/main/java/edu/wpi/first/wpilibj/DataLogManager.java#L238-L270

Should this include a warning when deleting logs because by the looks of it, it does not seem to inform anyone that it has hit the free space threshold and is running out of storage which would lead to logs being deleted without teams knowing.

Gold856, 2023-05-29:
Can't reproduce in simulation. I filled up a 16 GB flash drive with garbage data (to get to 50 MB of free space faster) and restarted robot code until I hit 11 logs, at which point, the oldest log was deleted. I kept restarting it, and it always maintained 11 logs after the program stopped. There are some edge cases where this could happen, like if you somehow have <10 logs, and your next log fills up the remaining free space, the oldest log would not get deleted. That's probably worth its own fix, but it's possible that, maybe during some testing period, a HUGE log was created (possibly because the robot was just on for several minutes,) and because there was less than 10 logs, the code never deletes the oldest log. Additionally, if for some reason the file can't be deleted, the data log manager will still create more logs. That could also lead to logs not getting deleted properly, and filling up the whole disk.

This is a bit of a tangent, but is there any way we could unit test this behavior? We probably can't do it as a Gradle task or CI (since you need to fill a partition so full as to have less than 50 MB of free space,) but can we create a VM image or something to run that can do this instead?

closed 2023-10-01 by PeterJohnson

TheTripleV, 2023-10-24:
Got a repro (On a rio2 running wpilib 2023.4.3 / robotpy 2023.4.3.0)

When the issue occurred, the robot no longer has robot code and robot code will stay permanently red.

The robotpy deploy logs results in
```
21:30:10:399 INFO    : paramiko.transport  : Connected (version 2.0, client OpenSSH_8.3)
21:30:10:520 INFO    : paramiko.transport  : Auth banner: b'NI Linux Real-Time (run mode)\n\nLog in with your NI-Auth credentials.\n\n'
21:30:10:520 INFO    : paramiko.transport  : Authentication (password) successful!
RoboRIO has WPILib version 2023.4.2.0
ERROR: configuring command: Command 'echo "env LD_LIBRARY_PATH=/usr/local/frc/lib/ /usr/local/bin/python3 -u -O /home/lvuser/py/robot.py run" > /home/lvuser/robotCommand' returned non-zero error status 1
[phoenix-diagnostics] Server shutdown cleanly. (dur:0)
[phoenix] Library shutdown cleanly
```
Normally files would be copied over at that point.

The rio has 2.86GB of log files, ranging from 0B in size all the way up to 113 MB. There's ~20 0B log files for some reason.

The logs are zipped in the file below.

https://drive.google.com/file/d/1_fVRyvhCPGZ8DJPmzpEfwpv1ThsctIxj/view?usp=sharing