wpilibsuite/allwpilib #5284: DutyCycleEncoder has 1-2 degree variance every second for CTRE magnetic encoder

calcmogul, 2023-04-24:
**Describe the bug**
The encoder should have a resolution of 360°/2¹⁴ = 0.02°, but it's more like 1.4° in practice.

Here's the FRC Discord traffic about it:
https://discord.com/channels/176186766946992128/368993897495527424/1100097745198522408

> thadhouse WPILib Alumni — Today at 9:50 AM
Sampling is 1mhz
How it works is actually stupid simple. It starts counting every microsecond when the signal goes high, and then the output is the number of counts when the output goes low.
And that’s just repeated.
If you’re using getOutput, youll get a dancing signal due to the way frequency is computed.
You have to use GetHighTime to not have that frequency wobble. Also DutyCycleEncoder will always be affected by frequency.
Only the raw DutyCycle class allows you to bypass the frequency calculation.

The suggested fix is to use DutyCycle.getHighTime() instead of DutyCycle.getFrequency(), but ideally we'd have a DutyCycleEncoder class without this problem. Apparently, the problem with using getHighTime() instead of getFrequency() is the position measurement and rollover calculation can get out of sync.

Note that this issue doesn't occur with REV through-bore due to it using a higher frequency. Its resolution is 360°/2¹⁰.

KangarooKoala, 2025-07-17:
This is closed by #8076, right? (Not sure why the autolinking didn't work)

calcmogul, 2025-07-17:
Closed by #8076.

PR #8076 (2025-07-15): Add frequency support to DutyCycle
Makes #8040 obsolete.

Closes #5284
closed 2025-07-17 by calcmogul