wpilibsuite/allwpilib #5302: Refactor existing WPILib commands to support suppliers

AngleSideAngle, 2023-04-30:
**Is your feature request related to a problem? Please describe.**
It seems the best way to handle state at runtime with command factories is with suppliers being fed into commands. However, `TrapezoidProfileCommand`, for example, doesn't support this. This means it has to be wrapped in a `DeferredCommand`, leading to a lot of annoying complexity.

**Describe the solution you'd like**
I think the simplest solution to this issue is to replace state with suppliers in the various classes that need it. To continue supporting existing functionality, they could have alternate constructors that set the supplier to supply the arguments.

Also see #5150 for deferred commands, and the pitfall surrounding using `ProxyCommand` for this.

Gold856, 2023-07-30:
What other commands need to be refactored to use suppliers? #5457 already did that for `TrapezoidProfileCommand`.

AngleSideAngle, 2023-07-30:
I forget exactly what my thought process was when I created this issue, but trapezoid profile command seems like the only command that needs it.

closed 2023-07-30 by AngleSideAngle