wpilibsuite/allwpilib #5307: Rename Speed objects to Velocity

srimanachanta, 2023-05-01:
Currently, all the data objects that track velocities are named `<>Speeds`. This is misleading as the objects track velocities and not speeds. They should be renamed to reflect that. The following refactors would need to be made:
- `ChassisSpeeds` -> `ChassisVelocities`
- `DifferentialDriveWheelSpeeds` -> `DifferentialDriveWheelVelocities`
- `MecanumDriveWheelSpeeds` -> `MecanumDriveWheelVelocities`
- `DifferentialDrive::WheelSpeeds` -> `DifferentialDrive::WheelVelocities`
- `MecanumDrive::WheelSpeeds` -> `MecanumDrive::WheelVelocities`

sciencewhiz, 2023-05-01:
If this change were made, could the user code be updated automatically by project importer? If not, I'm not sure the rename is worth the trouble, given that the docs all talk about velocity. 

https://github.com/wpilibsuite/vscode-wpilib/blob/main/vscode-wpilib/resources/java_replacements.json

rzblue, 2023-05-01:
Even if the project importer can fix the names, this seems like a needlessly breaking change, right after there were lots of complaints about breaking changes.

prateekma, 2023-05-02:
I don't think we need to be this pedantic here. The API docs clearly communicate the intent and I find it hard to believe that someone would actually be confused about what the classes do after reading the API docs / class contents. 

It's not worth it to increase the character count and to break teams over this.

rzblue, 2023-06-02:
Can this be closed?

closed 2023-06-02 by srimanachanta