wpilibsuite/allwpilib #5311: Make a JNI for Pose3d.exp() and Pose3d.log()

calcmogul, 2023-05-05:
**Is your feature request related to a problem? Please describe.**
The quoted post in https://www.chiefdelphi.com/t/2023-wpilib-feedback/435709/11 mentions Pose3d.exp() and Pose3d.log() taking up a significant portion of the loop time. It could be due to allocations and GC traffic from EJML.

**Describe the solution you'd like**
We should try adding a JNI for them that calls the C++ version and compare the performance between the two implementations. Here's the relevant files to modify:

* wpimath/src/main/java/edu/wpi/first/math/WPIMathJNI.java
* wpimath/src/main/native/cpp/jni/WPIMathJNI.cpp
* wpimath/src/test/java/edu/wpi/first/math/WPIMathJNITest.java

prateekma, 2023-05-05:
Duplicate of #5285?

calcmogul, 2023-05-05:
Yea, though this one has more information.

PR #5444 (2023-07-18): Add JNI for Pose3d exp and log
Fixes #5311

Just adds the JNI methods and tests which check that they work the same as the pure Java implementations, as requested in the issue. (I don't know how to compare the performance, though)
closed 2023-07-18 by PeterJohnson