wpilibsuite/allwpilib #5318: NT3/NT4 interoperability

chauser, 2023-05-08:
**Describe the bug**
I'm not sure that these are bugs, but they are gotchas if you don't know about them and I haven't found them mentioned anywhere. I struggled with them while working to get the SmartDashboard app updated to use the latest NT APIs and I thought it worth documenting them. Some of the assertions here are speculation about how things work -- the observed behavior is consistent with the the speculations but it might actually work differently than what I speculate.
These behavior were observed with WPILib java robot code (in simulation) as the server, various versions of the code in my SmartDashboard PR as one of the clients and OutlineViewer running both NT3 and NT4 as another client to observe properties of entries as seen by the different client versions.
1. `long`/`double` confusion: `long` entries published by the robot code are seen as `double` by NT3 clients. NT3 clients cannot, therefore, publish to the same entry (type mismatch). If the NT3 client code publishes a `long`, it is seen on the robot as a `double` so robot code that expects to publish a `long` to the same key fails to do so (type mismatch). This behavior was observed in SmartDashboard's Scheduler widget where the Cancel buttons did not work when running as an NT3 client. Speculations: the NT3 network protocol only supports `double` so `long` values published through the API are transmitted as `double`. If this is the case, the observed behavior is expected. Is there somewhere to learn about such subtleties?
2. Unpublish'ing persistent entries: this problem begins with the observation that even in NT4, in order to actually remove a persistent entry via operations on a client, it must first be set not persistent -- this may similarly apply to retained entries. (I didn't find this documented but it makes sense.) The weirdness with NT3 is that if the client, through the  API, sets an entry to not persistent it changes to retained and subsequent unpublish'ing doesn't remove it. If the NT3 client, after first setting the entry to not persistent then sets the entry to not retained, subsequent unpublish'ing removes it. I don't understand this at all though I did find a comment at one place in the NT source code saying something like "treat all NT3 published entries as retained". Again I'm not sure where to look to find discussion of such subtle behaviors. 

**Desktop (please complete the following information):**
 - WPILib Version: 2023.4.3-1-g663703d -- as reported by the `ResourceInformation.json` files in the `build/` folder.

I think the conclusion I draw from this is that NT3 clients should be transitioned to NT4, though I understand there are some other performance issues that may point in the other direction at this moment.

PeterJohnson, 2023-05-08:
Essentially none of the NT3 compatibility approach is documented; it's all implementation defined behaviors.  The NT4 spec does not define how NT3 things map to NT4, and does not require either clients or servers be backwards compatible.  The idea was for the NT3 piece to be a temporary backwards compatibility layer/shim to the NT3 library for a couple of years to let the ecosystem transition.  The long term intent is for all clients and servers to be NT4.  In general, you will see quite poor/weird behavior trying to use the newer NT4 implementation and API in NT3 mode to a NT4 server, this is not recommended as the new API exposes features that the NT3 protocol does not have.

1. Yes, the NT3 protocol only supports `double`.  Support for other numeric types (`float` and `long`) were added in NT4.  Numeric conversions are performed (so you can call get/setInteger even if the topic is actually `double`), but it doesn't allow mismatched publishing.

2. Yes, all NT3 published entries are automatically marked as retained.  The reason for this is because of things like auto mode setting, where if the NT3 client disconnects, you still want to retain the value, and there's no way for NT3 clients to actually communicate that intent.  The NT3 protocol has a "delete" entry feature (which NT4 does not) which I believe is mapped to unpublishing on the compatibility layer, but the NT4 API does not provide a delete API.

chauser, 2023-05-09:
I'm going to close this since it's not really buggy behavior. I think just having an issue about it and Peter's answer will help other people understand better what they're seeing.

closed 2023-05-09 by chauser