wpilibsuite/allwpilib #5323: Run/Cancel buttons on Commands placed on SmartDashboard with putData do not work reliably using NT4

chauser, 2023-05-10:
**Describe the bug**
Early in the season we noticed that the Run and Cancel buttons on Commands placed on SmartDashboard would occasionally fail to work. It was frequent enough that we gave up using this way of testing commands as it was pretty scary when Cancel didn't work, especially on a Command that was just being tested. I think the frequency of it not working is around 1 in 5 and sometimes there are multiple failures in a row.

I finally got back to looking into this. Having spent a long time looking at internal workings of NetworkTables I'm stuck so I'm posting here what I have found which may help someone with more familiarity with the NT code figure out what's happening.

**To Reproduce**
I did this with the robot code in Java but from what I've observed I suspect the problem is actually at the NT level and thus would affect C++ code as well.
Use SmartDashboard.putData to put a Command on NetworkTables in some robot code that you can simulate on the desktop.
Connect a dashboard (glass is probably best) running on the same computer to the simulation with NT4. Connect an OutlineViewer as well (also with NT4 so you can see the details of the publication).  I used a command that changed behavior of some LEDs so I could easily see the change in behavior when the command was running and not. 
Observe that on glass when you push the Run/Cancel buttons sometimes they don't work. On glass what I see is the display briefly changes to the expected state and then returns to the prior state. On the simulation desktop and outline viewer the state never changes. Furthermore, the timestamp on outline viewer does not change.
Observe that if you switch glass to an NT3 connection the problem does not occur.

**Expected behavior**
When Run is pressed the command enters running state. When Cancel is pressed the command stops running.

**Desktop (please complete the following information):**
 - WPILib Version: 2023.4.3
 - OS: Linux; also observed with dashboard running on Windows and Robot code running on the roborio (i.e. I don't believe the problem is OS-specific).
 - Java version whatever ships with the WPILib installer
 - C++ version -- from WPILib installer
 - 
**Additional context**
I ran wireshark and observed packets on port 5810: pressing one of the buttons immediately results in a packet which triggers a reply from the server. I suspect that the client for the dashboard is initially doing the state change and the reply packet from the server is what is changing it back.

I set a logging breakpoint on the code in CommandBase that changes the state of the command. When a button press does not work the setter is not called (and when it does work the breakpoint is triggered so I know the breakpoint was working).

I tested for the same behavior with SendableChooser -- it did not occur. SendableChooser does not provide a getter argument for initSendable.

I tested with a simpler SendableBoolean class that just sets the value of a member boolean variable in the setter argument of initSendable and gets it in the getter argument. The problem DID occur in this test.

I suspect a race between the NT update being done by SmartDashboard update based on the value returned by the getter and the incoming change from the network. But I'm stuck figuring out where in the NT code that could occur in a way that would leave the timestamp on the entry unchanged and why the behavior would be different for NT3- and NT4-connected dashboards.

Note that I'm pretty sure that all this is not related to time synchronization between the NT client and server. The round-trip-times for localhost should be small and consistent. Rather, I think the issue is related to the arrival time of the update message at the server and how that relates to the timing of the SmartDashboard update processing.

sciencewhiz, 2023-05-10:
Potentially same root cause as https://github.com/wpilibsuite/shuffleboard/issues/768

chauser, 2023-05-10:
I agree that this very well could be the cause of shuffleboard#768. 

I've now made a very basic C++ robot project with a C++ version of my SendableBoolean class. It exhibits the same problem; as in the java code, when the remote update fails the setter method of the SendableBoolean is not called.

chauser, 2023-05-12:
After figuring out how to turn on debug logging, I was able to discover the trigger event for the misbehavior. Updates are lost when the client's (dashboard's) incoming value at the server is time-stamped earlier than a previous value update pushed to NT by the SmartDashboard update loop. When a Sendable has a getter() function attached the getter is called on every update loop, so the timestamp in NT's LocalStorage advances when it processes that value. The incoming value with the earlier timestamp is ignored.
It's a bit surprising to me that this happens with the frequency that it does when the client is on the same machine as the server; but the incoming update is typically only "late" by a few hundred microseconds to a couple of milliseconds so it seems possible that process scheduling vagaries could cause this. When the client is on a different machine I would expect the occurrence to be more frequent.
It is not clear to me what to do to fix this. The problem doesn't occur using NT3. I conjecture that the NT3 code is assigning the timestamp at the time the message arrives on the server which would considerably narrow the circumstances where this kind of problem could occur. I don't know what use-cases we have that call for client-assigned timestamps but that is working against us in the Command Sendable's current design.
I wonder if a design that uses two fields -- one for the current state, set only by the robot program, and one for the desired change, set only by the dashboard program, would work. This is the approach used in the CommandScheduler's Sendable instance. 
Meanwhile, I would not recommend that anyone put commands on a dashboard and expect to control them reliably using the Run/Cancel buttons there. Documentation encouraging this usage should indicate that the feature is currently broken (at least when using NT4).

PeterJohnson, 2023-05-12:
I think we should change the approach so it’s not reliant on timing between multiple writers.  Fundamentally it’s a race condition and always has been.  We should separate it into control (written by dashboard) and status (written by robot) topics.

chauser, 2023-05-12:
Yes. The CommandScheduler's Sendable uses that approach. It's unfortunately the case that such a change affects all of the dashboard programs, but I think biting the bullet and getting it done is necessary in order to have commands controllable from the dashboards.
I wonder if there are other Sendables that exhibit the same problem? It shouldn't be too hard to figure out as any that have both a setter and getter are at least suspect.

chauser, 2023-05-13:
There are 13 calls to initSendable with both a getter and a setter in wpilibj (and presumably in wpilibc). This doesn't count any that might be in vendor libraries. 

My first thought on how to approach this was to modify SendableBuilderImpl to create a "control" NT entry along with the "value" entry for any Property that has a setter. I think that would work, but it has the downside that all dashboards need to be updated -- and depending on how they are implemented it might require changes for multiple widgets for each dashboard. 

@PeterJohnson's "If the robot code is constantly writing the value" phrase, however, triggered the thought: "what if the robot code does NOT constantly write the value [to NT]?" What if the Property implementation suppresses duplicate writes by either reading the current value from the NT entry or by keeping a cached local copy to compare against? Would that work? Would it have downsides? I'm not seeing any? As for performance, I think that a get operation on an NT entry in each update cycle is probably somewhat less costly than the current put operation that occurs with the same frequency. 

This approach has the virtue that the protocol for interacting with dashboards doesn't change so the dashboard programs don't need to change.

Starlight220, 2023-05-13:
Perhaps the network schedule/cancel functionality should be delegated to NetworkButton, rather than an in-place impl?

chauser, 2023-05-13:
@Starlight220 is that the right number? I don't see the relevance.

sciencewhiz, 2023-05-13:
Should be #5179

chauser, 2023-07-24:
PR #5356 (now merged) should have fixed this, but it should be tested with allwpilib HEAD to make sure. I'm not able to do that.

closed 2023-07-24 by chauser