wpilibsuite/allwpilib #5327: Unexpected ParallelRaceGroup interrupt behavior

NoahStricker, 2023-05-14:
Upon ending, a `ParallelRaceGroup` will call the `end()` method of the contained commands with the `interrupted` parameter set based on if the command group completed or was interrupted. However, when one of the internal commands in the command group ends, causing the command group to end, all of the commands within the command group are ended with `interrupted = false`, even if these commands were not yet completed and therefore should have ended with `interrupted = true`.

Something that should be made sure of here when fixing this is that if multiple commands finish on the same loop cycle, those commands should be ended with `interrupted = false`. A potential way to fix this is to have a hashmap of commands to completion status like `ParellelCommandGroup`, where the `isFinished()` method of all commands is checked, and if any are finished, those that returned `isFinished()` `true` should end with `interrupted = false`, while all others are ended with `interrupted = true`.

Starlight220, 2023-05-14:
I'm not sure what's causing that behavior, as this is the impl:

https://github.com/wpilibsuite/allwpilib/blob/main/wpilibNewCommands/src/main/java/edu/wpi/first/wpilibj2/command/ParallelRaceGroup.java#L81-L85

NoahStricker, 2023-05-14:
@Starlight220 looks like I was mistaken, I had thought that line 83 called the command groups isFinished method, whereas it actually calls the command's isFinished. I will close the issus now.

closed 2023-05-14 by NoahStricker