wpilibsuite/allwpilib #5337: Making Custom Widgets

narmstro2020, 2023-05-16:
What happened to the code for making custom widgets.  The code in the documentation doesn't work anymore.  Was this pulled.  Is there a new way to make custom widgets.  If so how.

calcmogul, 2023-05-16:
Wouldn't widget UI-related stuff be in https://github.com/wpilibsuite/shuffleboard/?

sciencewhiz, 2023-05-16:
I'm not sure what documentation doesn't work any more. Can you be more specific?

If it's [the frc-docs shuffleboard custom widget docs](https://docs.wpilib.org/en/stable/docs/software/dashboards/shuffleboard/custom-widgets/index.html) those were updated prior to this season in https://github.com/wpilibsuite/frc-docs/pull/2014 any issues with those docs should be put in the frc-docs repository.

narmstro2020, 2023-05-16:
So I started to create the sample plug-in at the top of this page in my project, but none of the imports would resolve.Sent from my iPhone

If it's the frc-docs shuffleboard custom widget docs those were updated prior to this season in wpilibsuite/frc-docs#2014 any issues with those docs should be put in the frc-docs repository.

narmstro2020, 2023-05-16:
Am I missing something.  I've got a blank robot project and started a blank class file from the wpilib docs.  But no imports are resolving.

narmstro2020, 2023-05-16:
So for example.  And sorry I forgot to mention those are the docs mentioned above.  

This import 

import edu.wpi.first.shuffleboard.api.data.ComplexData;

does not resolve.

calcmogul, 2023-05-16:
Does it build though? Intellisense can lie sometimes. The class exists at https://github.com/wpilibsuite/shuffleboard/blob/main/api/src/main/java/edu/wpi/first/shuffleboard/api/data/ComplexData.java.

narmstro2020, 2023-05-16:
So just to document my steps.
1.  Creating a test robot project.  
2. Created a java file called test plugin
3. Copied the sample plugin code at the top of https://docs.wpilib.org/en/stable/docs/software/dashboards/shuffleboard/custom-widgets/creating-plugins.html
4. imports not recognizing
5. Ran build
6. Did not build

calcmogul, 2023-05-16:
The docs say to use a subdirectory of the shuffleboard repo for the build, not a robot project.
https://docs.wpilib.org/en/stable/docs/software/dashboards/shuffleboard/custom-widgets/creating-plugins.html#building-plugin

closed 2023-08-02 by calcmogul

narmstro2020, 2023-08-17:
Hello.  I was wondering if this issue was fixed and if there was a build that incorporates it that I can apply to my testing.  Sent from my iPhone

Closed #5337 as completed.

calcmogul, 2023-08-17:
We tagged this issue as "support" because it was about proper usage of existing functionality. It's not clear that there was any bug in need of fixing. As I stated before, you should be using the shuffleboard repo to build plugins.

narmstro2020, 2023-08-17:
I’m sorry for the mixup on the reply email.  I couldn’t find the one for 5339.  I was wondering if 5339 was fixedSent from my iPhone

calcmogul, 2023-08-17:
Please leave a comment on https://github.com/wpilibsuite/allwpilib/issues/5339 instead for tracking purposes.