wpilibsuite/allwpilib #5339: Elevator Space Simulation Example Bug

narmstro2020, 2023-05-17:
So I've been playing around with LQR control and the Elevator State Space Simulation Example at: 

https://github.com/wpilibsuite/allwpilib/tree/main/wpilibjExamples/src/main/java/edu/wpi/first/wpilibj/examples/statespaceelevator

I'm just using the example code and I've noticed that if I change the gear ratio from 6 to 16 the simulator crashing with the following state trace and error 

Error at frc.robot.Robot.<init>(Robot.java:65): Unhandled exception instantiating robot edu.wpi.first.math.WPIMathJNI java.lang.RuntimeException: Failure at D:\a\allwpilib\allwpilib\wpimath\src\main\native\thirdparty\drake\src\math\discrete_algebraic_riccati_equation.cpp:420 in DiscreteAlgebraicRiccatiEquation(): condition 'es.eigenvalues()[i] >= 0' failed.
        at edu.wpi.first.math.WPIMathJNI.discreteAlgebraicRiccatiEquation(Native Method)
        at edu.wpi.first.math.Drake.discreteAlgebraicRiccatiEquation(Drake.java:24)
        at edu.wpi.first.math.Drake.discreteAlgebraicRiccatiEquation(Drake.java:53)
        at edu.wpi.first.math.estimator.KalmanFilter.<init>(KalmanFilter.java:92)
        at frc.robot.Robot.<init>(Robot.java:65)
        at edu.wpi.first.wpilibj.RobotBase.runRobot(RobotBase.java:298)
        at edu.wpi.first.wpilibj.RobotBase.lambda$startRobot$0(RobotBase.java:422)
        at java.base/java.lang.Thread.run(Thread.java:833)

calcmogul, 2023-05-17:
Since that's coming from KalmanFilter, that message can happen if Q or R have negative eigenvalues (i.e., the standard deviations used to construct them are less than zero).

If you sync with the latest main, we replaced the DARE solver and should give a more useful exception message.

narmstro2020, 2023-05-17:
Do I need to update the vendor library to do this.  It doesn't show any updates available.

calcmogul, 2023-05-17:
Oh, since you linked to allwpilib main, I figured you were building the example directly in the monorepo. I'll make the gear ratio change you mentioned and try to reproduce the exception.

narmstro2020, 2023-05-17:
okay. Sorry about that.  I'm just building a robot the normal way :) and didn't know where else to post.

calcmogul, 2023-05-17:
I reproduced it on main:
```
terminate called after throwing an instance of 'std::invalid_argument'
  what():  Q isn't positive semidefinite!

Q =
  0.000010  0.028286
  0.028286  0.000000
```

narmstro2020, 2023-05-18:
So I'm not crazy then.  A bug in the calculation

calcmogul, 2023-05-18:
Discretization.discretizeAQTaylor() doesn't match Discretization.discretizeAQ(), and it gets less accurate as more terms are added to the Taylor series expansion. Switching to discretizeAQ() fixes it. The former is less computationally expensive, so I'll look into fixing it.

narmstro2020, 2023-05-18:
I wonder if this bug exists in the Arm and other Linear systems

calcmogul, 2023-05-18:
It can affect them all, in theory.

calcmogul, 2023-05-18:
```
A_c =
Type = DDRM , rows = 2 , cols = 2
 0.0000E+00  1.0000E+00 
 0.0000E+00 -1.3479E+03 

Q_c =
Type = DDRM , rows = 2 , cols = 2
 2.5806E-03  0.0000E+00 
 0.0000E+00  1.0323E+00 

Taylor Q_d =
Type = DDRM , rows = 2 , cols = 2
 9.6533E-06  2.8286E-02 
 2.8286E-02  1.8302E-10 

Exact Q_d =
Type = DDRM , rows = 2 , cols = 2
 5.1624E-05  2.8411E-07 
 2.8411E-07  3.8291E-04
```
The Taylor series uses 6 terms. Adding more terms makes it diverge from the true value.

calcmogul, 2023-07-15:
Instead of fixing Discretization.discretizeAQTaylor(), we should remove it and modify Discretization.discretizeAQ() to use an algorithm that exploits the structure of the matrix exponential better.

A viable approach may be using a modified version of the Padé approximant Eigen uses.

narmstro2020, 2023-08-17:
Sorry for the confusion earlier.  
Just wanted to see if this was fixed in the current version and how can I build to it.

calcmogul, 2023-08-17:
The behavior you saw was fixed in https://github.com/wpilibsuite/allwpilib/pull/5344, which is not in any published release. The underlying issue has not been fixed yet, which is why this issue is still open.

See https://github.com/wpilibsuite/allwpilib/blob/main/DevelopmentBuilds.md for how to use development builds, which have the fix.

narmstro2020, 2023-08-17:
Okay I tried to run with the latest build and with
LinearSystemId.createpositionSystem I'm getting some errors with class
loaders.  I've followed the instruction to the letter on development
builds,

Here is the last stack trace

NT: Listening on NT3 port 1735, NT4 port 5810
Error at
java.base/jdk.internal.loader.BuiltinClassLoader.loadClass(BuiltinClassLoader.java:641):
Unhandled exception: java.lang.ClassNotFoundException:
org.ejml.simple.ConstMatrix
        at
java.base/jdk.internal.loader.BuiltinClassLoader.loadClass(BuiltinClassLoader.java:641)
        at
java.base/jdk.internal.loader.ClassLoaders$AppClassLoader.loadClass(ClassLoaders.java:188)
        at java.base/java.lang.ClassLoader.loadClass(ClassLoader.java:520)
        at
edu.wpi.first.math.system.plant.LinearSystemId.identifyPositionSystem(LinearSystemId.java:269)

narmstro2020, 2023-08-17:
Never mind I needed to add the org.ejml dependency

PR #5562 (2023-08-23): [wpimath] Remove discretizeAQTaylor()
It gives incorrect results. Any replacement should just be an implementation detail of discretizeAQ().

Closes #5339.
commit 8f3d6a1d (2023-08-23): [wpimath] Remove discretizeAQTaylor() (#5562)

It gives incorrect results. Any replacement should just be an
implementation detail of discretizeAQ().

Closes #5339.
closed 2023-08-23 by PeterJohnson