wpilibsuite/allwpilib #5350: Add a `withinTolerance` method to `MathUtil`

Gold856, 2023-05-19:
**Is your feature request related to a problem? Please describe.**
It's useful to check if a mechanism is at certain point, e.g., checking if an arm is at a certain position. Unfortunately, the real world is difficult and mostly doesn't work with exact integers, so motors almost never stop at exact integer degrees.

**Describe the solution you'd like**
Add a method to `MathUtil` with parameters for a current value, target value, and tolerance. Overload it with additional parameters for a minimum value and a maximum value to account for things like absolute encoders.

**Describe alternatives you've considered**
Letting teams figure out tolerance checking by themselves. However, I think most teams would find a tolerance checking function useful.

**Additional context**
I have some code I wrote to do tolerance checking, so I can submit a PR with it, but I wanted to discuss it first.

calcmogul, 2023-05-19:
You can use absolute value for that.
```java
if (Math.abs(target - actual) < tolerance) {
```

Gold856, 2023-05-19:
But having 5 parameters for a method seems kind of excessive

Gold856, 2023-05-19:
The code I would like in a `withinTolerance` or an `isNear` method(which I'll submit as a PR if desired) also supports continuous input like this(this checks if an encoder is within a certain number of degrees with continuous input support):

	double upperAngleBound = targetAngle + tolerance;
	double lowerAngleBound = targetAngle - tolerance;
	if (Math.abs(currentAngle - targetAngle) < tolerance) {
		return true;
	} else if (lowerAngleBound < 0 || upperAngleBound > 360) {
		double wrappedLowerAngleBound = MathUtil.inputModulus(lowerAngleBound, 0, 360);
		double wrappedUpperAngleBound = MathUtil.inputModulus(upperAngleBound, 0, 360);
		if (currentAngle > wrappedLowerAngleBound || currentAngle < wrappedUpperAngleBound) {
			return true;
		}
	}
	return false;

Doing tolerance checks without continuous input support is easy, but it's a little bit more complicated with continuous input support, which is why I would like a method to do this in `MathUtil`.

calcmogul, 2023-05-19:
Here's what PIDController does for that:
```java
  double errorBound = (m_maximumInput - m_minimumInput) / 2.0;
  m_positionError = MathUtil.inputModulus(m_setpoint - m_measurement, -errorBound, errorBound);
```

So a function for it would look like:
```java
boolean isNear(double expected, double actual, double tolerance, double min, double max) {
  double errorBound = (max - min) / 2.0;
  double error = MathUtil.inputModulus(expected - actual, -errorBound, errorBound);
  return Math.abs(error) < tolerance;
}
```

Gold856, 2023-05-19:
That's... a lot simpler than my implementation. Can I create a PR with your code?

calcmogul, 2023-05-19:
Go ahead. Don't forget to translate it to C++.

Gold856, 2023-05-19:
If the error equals the tolerance, should that return true?

Gold856, 2023-05-19:
Also, we should probably disallow negative tolerances.

calcmogul, 2023-05-19:
> If the error equals the tolerance, should that return true?

All our controller functions like atSetpoint() return true when the error is < setpoint instead of <= setpoint, so just do `<`.

>  we should probably disallow negative tolerances

That's already an implicit assumption basically everywhere in WPILib.

Gold856, 2023-05-20:
The C++ compiler is very unhappy at me for mixing ints and doubles in the tests. How do I fix that?

calcmogul, 2023-05-20:
I'd have to see the error messages.

Gold856, 2023-05-20:
https://pastebin.com/ycXrSHtT

calcmogul, 2023-05-20:
If T is ambiguous, you have to specify it manually via `frc::IsNear<TypeHere>(`.

Gold856, 2023-05-20:
I don't really know C++, how do I do that when I'm using both doubles and ints?

calcmogul, 2023-05-20:
Force doubles with `frc::IsNear<double>(`

closed 2023-06-20 by PeterJohnson