wpilibsuite/allwpilib #5368: Embedded Simulation Function in Classes

Weq511762338, 2023-05-31:
**Is your feature request related to a problem? Please describe.**
It is not any bug or problem, but I find writing the simulation code for every single motor/encoder in a large project unfriendly to beginners and hard to be kept track of. For example, blocks of code that update sim encoders, sim motors, or robot poses may appear in multiple places in a robot program. This can be quite inconvenient when one needs to locate all of these places when making changes, or when one wants to comment the simulation functionality out completely.

It is also beginner-unfriendly that one needs to learn all different ways to simulate different devices and understand when to update the sim devices in order to write the simulation code for a current project, especially when this project is complicated.

**Describe the solution you'd like**
Centralize, integrate all updating methods from the simulation devices into the methods from the actual devices by creating a generic class or wrapper (to handle different vendors) Create sim device objects inside the actual device classes.

For example, when setting the actual motor's voltage to 1, the sim device inside the actual motor class will update the encoder sim device for that motor object accordingly without having the user explicitly write out the lines.

Other beginner-friendly features can come along in this structure: since the sim devices are in the object of actual devices, we can choose to turn simulation for that device on or off freely, without going all over the place to comment out all the sim code. We can also freely choose to customize the simulation graphics, or choose simulations for which specific devices to be turned on/off.

**Describe alternatives you've considered**
Other forms of centralizing simulation

**Additional context**
I will put further details here within a week soon. I will first try to implement this structure with a simple motor device.

***😃Please ask any questions or for clarifications! English is not my first language so I am not sure I described it well***

PeterJohnson, 2023-05-31:
It's not possible for us to distribute a wrapper to handle different motor vendors. We build our library, then vendors build their library on top; we can't depend on their library as that would create a dependency loop.

There's a few benefits for the sim classes to be separate:
- It makes it clear what is actually run on the robot and what is run only in simulation
- It provides entry points for physics modeling. Note in order to generate an encoder value from a motor output, you have to provide some kind of physics model (e.g. what load is the motor driving?).
- It provides a separation of concerns and flexibility/entry points for modeling more complex behaviors than just a single motor with an integrated encoder.  E.g. what if you have two motors powering a gearbox with a pneumatic shifter and the encoder is after the shifter (relatively common in drivetrains)?  Also, not all motors have integrated encoders.

pjreiniger, 2023-06-02:
I have a [library](https://github.com/snobotsim/SnobotSimV2/tree/master) available by a [vendordep](https://github.com/GirlsOfSteelRobotics/GirlsOfSteelFRC/blob/main/y2023/ChargedUp/vendordeps/SnobotSim.json) which is like syntactic sugar around the the sim classes. It basically manages all of the member variables for you to remove a lot of the annoying boilerplate code. It also support simulating advanced REV features (although maybe not accurately) as well as wrappers for CTRE and NavX. Makes for a common interface instead of having to deal with the differences of `encoderSim.setPosition` for WPI things, `talonSrx.getSimCollection().setQuadratureRawPosition` for CTRE and `SimDeviceSim("navX-Sensor[0]").GetDouble("yaw").set()` for navx

Diff Drive example:
```
            m_simulator = new DifferentialDrivetrainSimWrapper(
                drivetrainSim,
                new RevMotorControllerSimWrapper(m_leaderLeft),
                new RevMotorControllerSimWrapper(m_leaderRight),
                RevEncoderSimWrapper.create(m_leaderLeft),
                RevEncoderSimWrapper.create(m_leaderRight),
                new CtrePigeonImuWrapper(m_gyro));
    }

    @Override
    public void simulationPeriodic() {
        m_simulator.update();
    }