wpilibsuite/allwpilib #5385: CoordinateSystem convert incorrect results for Transfom3d Java

tom131313, 2023-06-09:
The `Math Geometry CoordinateSystem` overloaded `convert` methods can convert a Pose3D or Transform3d.

The results are identical but only appropriate for the Pose3d which describes an orientation and are incorrect for the Transform3d which describes a delta (change in values).

The trivial example with all zeros for 3D transform and 3D rotation; EDN to NWU conversion:

origin pose3d converted
0.0 0.0 0.0; -90.0 0.0 -90.0

no change transform3d converted
0.0 0.0 0.0; -90.0 0.0 -90.0

This causes the transform to subsequently attempt a -90 degree rotation when none was desired.

In addition the converted transform rotation axis does not follow the correctly converted transform translation axis causing the the new incorrect delta rotation transform to be applied to the incorrect axis.
 [Tx -> -Ty; Rx -> Rx-90]
 [Ty -> -Tz; Ry -> -Rz-90]
 [Tz ->Tx; Rz -> Ry]

WPILib 2023.4.3
Problem discovered while attempting to process an AprilTag pose to robot using the WPILib examples and library methods for coordinate conversion from EDN to NWU.
```java
for(int axis = 0; axis<=2; axis++)
{
  double[] v = {0., 0., 0.};
  v[axis] = 10.;
  var tp = new Transform3d(
                new Translation3d(v[0], v[1], v[2]),
                new Rotation3d(
                    Units.degreesToRadians(v[0]),
                   Units.degreesToRadians(v[1]),
                    Units.degreesToRadians(v[2])));

  System.out.println("\ntp\n"
    + tp.getX() + " " + tp.getY() + " " + tp.getZ()
    + "; " + Units.radiansToDegrees(tp.getRotation().getX())
    + " " + Units.radiansToDegrees(tp.getRotation().getY())
    + " " + Units.radiansToDegrees(tp.getRotation().getZ())
    );

  var tpc = CoordinateSystem.convert(tp, CoordinateSystem.EDN(), CoordinateSystem.NWU());         

  System.out.println("tp converted\n"
    + tpc.getX() + " " + tpc.getY() + " " + tpc.getZ()
    + "; " + Units.radiansToDegrees(tpc.getRotation().getX())
    + " " + Units.radiansToDegrees(tpc.getRotation().getY())
    + " " + Units.radiansToDegrees(tpc.getRotation().getZ())
    );
}
var origin = CoordinateSystem.convert(new Pose3d(), CoordinateSystem.EDN(), CoordinateSystem.NWU());

System.out.println("\norigin pose3d converted\n"
  + origin.getX() + " " + origin.getY() + " " + origin.getZ()
  + "; " + Units.radiansToDegrees(origin.getRotation().getX())
  + " " + Units.radiansToDegrees(origin.getRotation().getY())
  + " " + Units.radiansToDegrees(origin.getRotation().getZ())
  );

var noChange = CoordinateSystem.convert(new Transform3d(), CoordinateSystem.EDN(), CoordinateSystem.NWU());

System.out.println("no change transform3d converted\n"
  + noChange.getX() + " " + noChange.getY() + " " + noChange.getZ()
  + "; " + Units.radiansToDegrees(noChange.getRotation().getX())
  + " " + Units.radiansToDegrees(noChange.getRotation().getY())
  + " " + Units.radiansToDegrees(noChange.getRotation().getZ())
  );

/*
tp
10.0 0.0 0.0; 10.0 0.0 0.0 [Tx -> -Ty; Rx -> Rx-90]
tp converted
0.0 -10.0 0.0; -80.00000000000001 0.0 -90.0

tp
0.0 10.0 0.0; 0.0 10.0 0.0 [Ty -> -Tz; Ry -> -Rz-90]
tp converted
0.0 0.0 -10.0; -90.0 0.0 -100.0

tp
0.0 0.0 10.0; 0.0 0.0 10.0 [Tz ->Tx; Rz -> Ry]
tp converted
10.0 0.0 0.0; -90.0 9.999999999999998 -90.0

origin pose3d converted
0.0 0.0 0.0; -90.0 0.0 -90.0
no change transform3d converted
0.0 0.0 0.0; -90.0 0.0 -90.0
 */
```

amquake, 2023-06-09:
This also confused me a while ago.

To illustrate the rotation problem, here is a hypothetical change in rotation in EDN:
 
[image]

When converting this rotation delta from EDN to NWU in wpilib, I expected to get this:

[image]

However, the `CoordinateSystem` conversion actually gives me something like this:

[image]

This is because the conversion finds the rotation to align the NWU origin axes exactly with the given rotation's EDN axes, instead of the rotation delta from the respective origin's axes.  

I ended up using this janky fix to get my expected behavior:
```java
    /**
     * Convert a rotation from EDN to NWU. For example, if you have a rotation X,Y,Z {1, 0, 0} in EDN,
     * this would be XYZ {0, -1, 0} in NWU.
     */
    private static Rotation3d rotationEDNtoNWU(Rotation3d rot) {
        return CoordinateSystem.convert(
                        new Rotation3d(), CoordinateSystem.NWU(), CoordinateSystem.EDN())
                .plus(CoordinateSystem.convert(rot, CoordinateSystem.EDN(), CoordinateSystem.NWU()));
    }
```

tom131313, 2023-06-10:
@amquake Nice! Thank you. I was thinking some sort of `minus` would help but I didn't have one; you cleverly implemented that with `plus` to the `reverse convert`.
I'll use this circumvention for now and it might be what WPILib should be but you and I have only used it on EDN to NWU:
```java
  public static Transform3d CoordinateSystemDOTconvert(
      Transform3d transform, CoordinateSystem from, CoordinateSystem to) {   
    return new Transform3d(
       CoordinateSystem.convert(transform.getTranslation(), from, to),
       CoordinateSystem.convert(new Rotation3d(), to, from)
                       .plus(CoordinateSystem.convert(transform.getRotation(), from, to)));
  }
```

PR #5532 (2023-08-12): [wpimath] Fix CoordinateSystem.convert() Transform3d overload
Fixes #5385 

The current rotation conversion finds the rotation which aligns the axes from the final coordinate system onto the axes in the initial coordinate system which were first rotated by the given `Rotation3d`. This is why converting an empty rotation returns a non-empty rotation-- for example, converting an empty rotation in EDN to NWU returns the rotation from the NWU axes to the EDN axes.

The expected behavior mentioned in the issue is for rotations to extrinsically "match" in both coordinate systems. The example of an e [... cut]
closed 2023-08-12 by PeterJohnson