wpilibsuite/allwpilib #5386: Use robot pose for isFinished in RamseteCommand, etc

cml1010101, 2023-06-09:
It is not the best solution that the RamseteCommand, etc uses the estimated time that a path would take to determine whether the command is finished, rather then check to see whether it is at the target position.

To fix this issue, there could be separate implementations, or a parameter. There could also be another class. Just some thoughts. So long as it checks the position of the robot relative

Starlight220, 2023-08-25:
This affects `TrapezoidProfileCommand` too.

The problem with this is that the robot may never reach the target position, and then the command will run infinitely and in an undefined way.

Gold856, 2024-06-18:
Resolved by #6494 since we're deprecating RamseteCommand.

closed 2024-06-18 by calcmogul