wpilibsuite/allwpilib #5423: Installation fails with `The SSL connection could not be established`

yabberyabber, 2023-07-06:
**Describe the bug**
When trying to install WPILib_Linux-2023.4.3 on ubuntu 22.04, I get the following error in the installer GUI:
```
System.Net.Http.HttpRequestException: The SSL connection could not be established, see inner exception.
 ---> System.Security.Authentication.AuthenticationException: The remote certificate is invalid because of errors in the certificate chain: NotTimeValid
   at System.Net.Security.SslStream.SendAuthResetSignal(ProtocolToken message, ExceptionDispatchInfo exception)
   at System.Net.Security.SslStream.CompleteHandshake(SslAuthenticationOptions sslAuthenticationOptions)
   at System.Net.Security.SslStream.ForceAuthenticationAsync[TIOAdapter](Boolean receiveFirst, Byte[] reAuthenticationData, CancellationToken cancellationToken)
   at System.Net.Http.ConnectHelper.EstablishSslConnectionAsync(SslClientAuthenticationOptions sslOptions, HttpRequestMessage request, Boolean async, Stream stream, CancellationToken cancellationToken)
   --- End of inner exception stack trace ---
   at System.Net.Http.ConnectHelper.EstablishSslConnectionAsync(SslClientAuthenticationOptions sslOptions, HttpRequestMessage request, Boolean async, Stream stream, CancellationToken cancellationToken)
   at System.Net.Http.HttpConnectionPool.ConnectAsync(HttpRequestMessage request, Boolean async, CancellationToken cancellationToken)
   at System.Net.Http.HttpConnectionPool.CreateHttp11ConnectionAsync(HttpRequestMessage request, Boolean async, CancellationToken cancellationToken)
   at System.Net.Http.HttpConnectionPool.AddHttp11ConnectionAsync(QueueItem queueItem)
   at System.Threading.Tasks.TaskCompletionSourceWithCancellation`1.WaitWithCancellationAsync(CancellationToken cancellationToken)
   at System.Net.Http.HttpConnectionPool.HttpConnectionWaiter`1.WaitForConnectionAsync(Boolean async, CancellationToken requestCancellationToken)
   at System.Net.Http.HttpConnectionPool.SendWithVersionDetectionAndRetryAsync(HttpRequestMessage request, Boolean async, Boolean doRequestAuth, CancellationToken cancellationToken)
   at System.Net.Http.RedirectHandler.SendAsync(HttpRequestMessage request, Boolean async, CancellationToken cancellationToken)
   at System.Net.Http.HttpClient.<SendAsync>g__Core|83_0(HttpRequestMessage request, HttpCompletionOption completionOption, CancellationTokenSource cts, Boolean disposeCts, CancellationTokenSource pendingRequestsCts, CancellationToken originalCancellationToken)
   at WPILibInstaller.Utils.HttpClientDownloadWithProgress.StartDownload() in /home/runner/work/WPILibInstaller-Avalonia/WPILibInstaller-Avalonia/WPILibInstaller-Avalonia/Utils/HttpClientDownloadWithProgress.cs:line 29
   at WPILibInstaller.ViewModels.VSCodePageViewModel.DownloadForPlatform(String downloadUrl, Stream outputStream, Action`1 progressChanged) in /home/runner/work/WPILibInstaller-Avalonia/WPILibInstaller-Avalonia/WPILibInstaller-Avalonia/ViewModels/VSCodePageViewModel.cs:line 383
   at WPILibInstaller.ViewModels.VSCodePageViewModel.DownloadToMemoryStream(Platform platform, String downloadUrl, Action`1 progressChanged) in /home/runner/work/WPILibInstaller-Avalonia/WPILibInstaller-Avalonia/WPILibInstaller-Avalonia/ViewModels/VSCodePageViewModel.cs:line 365
   at WPILibInstaller.ViewModels.VSCodePageViewModel.DownloadSingleVSCodeFunc() in /home/runner/work/WPILibInstaller-Avalonia/WPILibInstaller-Avalonia/WPILibInstaller-Avalonia/ViewModels/VSCodePageViewModel.cs:line 335
```

**To Reproduce**
Exactly as referenced here https://docs.wpilib.org/en/stable/docs/zero-to-robot/step-2/wpilib-setup.html
```
tar -xf WPILib_Linux-<version>.tar.gz

cd WPILib_Linux-<version>/

./WPILibInstaller
```

Then in the installer selecting `Download for this computer only (fastest)`

**Expected behavior**
I expected the provided installer to complete successfully

**Screenshots**
[image]

**Desktop (please complete the following information):**
 - OS: Ubuntu 22.04
 - WPILib Installer

ThadHouse, 2023-07-06:
Make sure your clock is right on your system. That error is saying theres a clock wrong somewhere.

yabberyabber, 2023-07-06:
I have NTP enabled on my machine so I expect my system clock can't be that far off

```
 > timedatectl 
               Local time: Thu 2023-07-06 15:00:44 PDT
           Universal time: Thu 2023-07-06 22:00:44 UTC
                 RTC time: Thu 2023-07-06 22:00:44
                Time zone: America/Los_Angeles (PDT, -0700)
System clock synchronized: yes
              NTP service: active
          RTC in local TZ: no
```
are there any other clocks involved here? It looks like the installer might be using the .net runtime --- is there a separate clock used by dotnet services maybe?

ThadHouse, 2023-07-06:
Hmm. Thats weird.

https://update.code.visualstudio.com/1.74.2/linux-x64/stable

Thats the file its trying to download, can you reach that manually?

yabberyabber, 2023-07-06:
yes my browser can download it just fine

ThadHouse, 2023-07-06:
Yeah, its a self contained binary, so the .NET Runtime is included in it.

sciencewhiz, 2023-07-06:
What type of internet connection?

yabberyabber, 2023-07-06:
> What type of internet connection?

IS there a specific attribute you have in mind?

I'm on my home wifi network. My ISP is charter. I've got about 375Mbps down and 20Mbps up

sciencewhiz, 2023-07-06:
The concern would a MITM proxy or firewall using certificates installed in the browser but not used by .net.

yabberyabber, 2023-07-07:
Ah that makes sense. I briefly considered whether my work VPN was causing an issue here but I disabled that pretty early on.

yabberyabber, 2023-07-13:
I was able to install it in a fresh docker 22.04 container on my laptop so whatever went wrong is probably very specific to my particular setup. I'll close the issue since it's unlikely that there's anything practically wrong with the installer itself.

closed 2023-07-13 by yabberyabber