wpilibsuite/allwpilib #5426: Failed to install wpilib

autoantwort, 2023-07-09:
Package: wpilib:arm64-osx -> main

**Host Environment**

- Host: arm64-osx
- Compiler: AppleClang 14.0.3.14030022
-    vcpkg-tool version: 2999-12-31-unknownhash
    vcpkg-scripts version: dafef74af 2023-07-07 (2 days ago)

**To Reproduce**

`vcpkg x-set-installed wpilib`

**Failure logs**

```
-- Downloading https://github.com/wpilibsuite/allwpilib/archive/d37f990ce3a8a36bc791dd989dcea2868759e556.tar.gz -> wpilibsuite-allwpilib-d37f990ce3a8a36bc791dd989dcea2868759e556.tar.gz...
-- Extracting source /Users/leanderSchulten/git_projekte/vcpkg/downloads/wpilibsuite-allwpilib-d37f990ce3a8a36bc791dd989dcea2868759e556.tar.gz
-- Using source at /Users/leanderSchulten/git_projekte/vcpkg/buildtrees/wpilib/src/868759e556-81ce2d8562.clean
-- Configuring arm64-osx
CMake Warning at vcpkg_installed/arm64-osx/share/vcpkg-cmake/vcpkg_cmake_configure.cmake:318 (message):
  The following variables are not used in CMakeLists.txt:

      FLAT_INSTALL_WPILIB

  Please recheck them and remove the unnecessary options from the
  `vcpkg_cmake_configure` call.

  If these options should still be passed for whatever reason, please use the
  `MAYBE_UNUSED_VARIABLES` argument.
Call Stack (most recent call first):
  ports/wpilib/portfile.cmake:18 (vcpkg_cmake_configure)
  scripts/ports.cmake:147 (include)

-- Building arm64-osx-dbg
CMake Error at scripts/cmake/vcpkg_execute_build_process.cmake:134 (message):
    Command failed: /Users/leanderSchulten/git_projekte/vcpkg/downloads/tools/cmake-3.25.1-osx/cmake-3.25.1-macos-universal/CMake.app/Contents/bin/cmake --build . --config Debug --target install -- -v -j11
    Working Directory: /Users/leanderSchulten/git_projekte/vcpkg/buildtrees/wpilib/arm64-osx-dbg
    See logs for more information:
      /Users/leanderSchulten/git_projekte/vcpkg/buildtrees/wpilib/install-arm64-osx-dbg-out.log
      /Users/leanderSchulten/git_projekte/vcpkg/buildtrees/wpilib/install-arm64-osx-dbg-err.log

Call Stack (most recent call first):
  vcpkg_installed/arm64-osx/share/vcpkg-cmake/vcpkg_cmake_build.cmake:74 (vcpkg_execute_build_process)
  vcpkg_installed/arm64-osx/share/vcpkg-cmake/vcpkg_cmake_install.cmake:16 (vcpkg_cmake_build)
  ports/wpilib/portfile.cmake:29 (vcpkg_cmake_install)
  scripts/ports.cmake:147 (include)

```
<details><summary>/Users/leanderSchulten/git_projekte/vcpkg/buildtrees/wpilib/install-arm64-osx-dbg-out.log</summary>

```
[1/267] cd /Users/leanderSchulten/git_projekte/vcpkg/buildtrees/wpilib/arm64-osx-dbg/wpiutil && /Users/leanderSchulten/git_projekte/vcpkg/downloads/tools/cmake-3.25.1-osx/cmake-3.25.1-macos-universal/CMake.app/Contents/bin/cmake -Dinput=/Users/leanderSchulten/git_projekte/vcpkg/buildtrees/wpilib/src/868759e556-81ce2d8562.clean/wpiutil/src/main/native/resources/wpilib-128.png -Doutput=generated/main/cpp/wpilib-128.png.cpp -Dprefix=WPI -Dnamespace=wpi -P /Users/leanderSchulten/git_projekte/vcpkg/buildtrees/wpilib/src/868759e556-81ce2d8562.clean/cmake/modules/../scripts/GenResource.cmake
[2/267] /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/c++  -I/Users/leanderSchulten/git_projekte/vcpkg/buildtrees/wpilib/src/868759e556-81ce2d8562.clean/wpiutil/src/main/native/thirdparty/fmtlib/include -I/Users/leanderSchulten/git_projekte/vcpkg/buildtrees/wpilib/src/868759e556-81ce2d8562.clean/wpiutil/src/main/native/include -I/Users/leanderSchulten/git_projekte/vcpkg/buildtrees/wpilib/src/868759e556-81ce2d8562.clean/wpiutil/src/main/native/thirdparty/memory/include -I/Users/leanderSchulten/git_projekte/vcpkg/buildtrees/wpilib/src/868759e556-81ce2d8562.clean/wpiutil/src/main/native/thirdparty/json/include -I/Users/leanderSchulten/git_projekte/vcpkg/buildtrees/wpilib/src/868759e556-81ce2d8562.clean/wpiutil/src/main/native/thirdparty/llvm/include -I/Users/leanderSchulten/git_projekte/vcpkg/buildtrees/wpilib/src/868759e556-81ce2d8562.clean/wpiutil/src/main/native/thirdparty/mpack/include -I/Users/leanderSchulten/git_projekte/vcpkg/buildtrees/wpilib/src/868759e556-81ce2d8562.clean/wpiutil/src/main/native/thirdparty/sigslot/include -fPIC -g -arch arm64 -isysroot /Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX13.3.sdk -Wall -pedantic -Wextra -Werror -Wno-unused-parameter -Wno-deprecated-anon-enum-enum-conversion -std=gnu++20 -MD -MT wpiutil/CMakeFiles/wpiutil.dir/src/main/native/cpp/timestamp.cpp.o -MF wpiutil/CMakeFiles/wpiutil.dir/src/main/native/cpp/timestamp.cpp.o.d -o wpiutil/CMakeFiles/wpiutil.dir/src/main/native/cpp/timestamp.cpp.o -c /Users/leanderSchulten/git_projekte/vcpkg/buildtrees/wpilib/src/868759e556-81ce2d8562.clean/wpiutil/src/main/native/cpp/timestamp.cpp
[3/267] /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/c++  -I/Users/leanderSchulten/git_projekte/vcpkg/buildtrees/wpilib/src/868759e556-81ce2d8562.clean/wpiutil/src/main/native/thirdparty/fmtlib/include -I/Users/leanderSchulten/git_projekte/vcpkg/buildtrees/wpilib/src/868759e556-81ce2d8562.clean/wpiutil/src/main/native/include -I/Users/leanderSchulten/git_projekte/vcpkg/buildtrees/wpilib/src/868759e556-81ce2d8562.clean/wpiutil/src/main/native/thirdparty/memory/include -I/Users/leanderSchulten/git_projekte/vcpkg/buildtrees/wpilib/src/868759e556-81ce2d8562.clean/wpiutil/src/main/native/thirdparty/json/include -I/Users/leanderSchulten/git_projekte/vcpkg/buildtrees/wpilib/src/868759e556-81ce2d8562.clean/wpiutil/src/main/native/thirdparty/llvm/include -I/Users/leanderSchulten/git_projekte/vcpkg/buildtrees/wpilib/src/868759e556-81ce2d8562.clean/wpiutil/src/main/native/thirdparty/mpack/include -I/Users/leanderSchulten/git_projekte/vcpkg/buildtrees/wpilib/src/868759e556-81ce2d8562.clean/wpiutil/src/main/native/thirdparty/sigslot/include -fPIC -g -arch arm64 -isysroot /Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX13.3.sdk -Wall -pedantic -Wextra -Werror -Wno-unused-parameter -Wno-deprecated-anon-enum-enum-conversion -std=gnu++20 -MD -MT wpiutil/CMakeFiles/wpiutil.dir/src/main/native/cpp/MappedFileRegion.cpp.o -MF wpiutil/CMakeFiles/wpiutil.dir/src/main/native/cpp/MappedFileRegion.cpp.o.d -o wpiutil/CMakeFiles/wpiutil.dir/src/main/native/cpp/MappedFileRegion.cpp.o -c /Users/leanderSchulten/git_projekte/vcpkg/buildtrees/wpilib/src/868759e556-81ce2d8562.clean/wpiutil/src/main/native/cpp/MappedFileRegion.cpp
...
Skipped 536 lines
...

-- Up-to-date: /Users/leanderSchulten/git_projekte/vcpkg/packages/wpilib_arm64-osx/debug/include/ntcore/networktables
-- Installing: /Users/leanderSchulten/git_projekte/vcpkg/packages/wpilib_arm64-osx/debug/include/ntcore/networktables/BooleanArrayTopic.h
-- Installing: /Users/leanderSchulten/git_projekte/vcpkg/packages/wpilib_arm64-osx/debug/include/ntcore/networktables/DoubleTopic.h
-- Installing: /Users/leanderSchulten/git_projekte/vcpkg/packages/wpilib_arm64-osx/debug/include/ntcore/networktables/RawTopic.h
-- Installing: /Users/leanderSchulten/git_projekte/vcpkg/packages/wpilib_arm64-osx/debug/include/ntcore/networktables/RawTopic.inc
-- Installing: /Users/leanderSchulten/git_projekte/vcpkg/packages/wpilib_arm64-osx/debug/include/ntcore/networktables/BooleanTopic.inc
-- Installing: /Users/leanderSchulten/git_projekte/vcpkg/packages/wpilib_arm64-osx/debug/include/ntcore/networktables/IntegerTopic.inc
-- Installing: /Users/leanderSchulten/git_projekte/vcpkg/packages/wpilib_arm64-osx/debug/include/ntcore/networktables/FloatArrayTopic.h
-- Installing: /Users/leanderSchulten/git_projekte/vcpkg/packages/wpilib_arm64-osx/debug/include/ntcore/networktables/FloatTopic.inc
-- Installing: /Users/leanderSchulten/git_projekte/vcpkg/packages/wpilib_arm64-osx/debug/include/ntcore/networktables/DoubleArrayTopic.inc
-- Installing: /Users/leanderSchulten/git_projekte/vcpkg/packages/wpilib_arm64-osx/debug/include/ntcore/networktables/IntegerArrayTopic.inc
[... 291 lines cut ...]
-- Installing: /Users/leanderSchulten/git_projekte/vcpkg/packages/wpilib_arm64-osx/debug/include/wpimath/Eigen/src/Eigenvalues/HessenbergDecomposition.h
-- Installing: /Users/leanderSchulten/git_projekte/vcpkg/packages/wpilib_arm64-osx/debug/include/wpimath/Eigen/src/Eigenvalues/ComplexEigenSolver.h
-- Installing: /Users/leanderSchulten/git_projekte/vcpkg/packages/wpilib_arm64-osx/debug/include/wpimath/Eigen/src/Eigenvalues/GeneralizedSelfAdjointEigenSolver.h
-- Installing: /Users/leanderSchulten/git_projekte/vcpkg/packages/wpilib_arm64-osx/debug/include/wpimath/Eigen/src/Eigenvalues/ComplexSchur.h
[... cut ...]
</details>
<details><summary>/Users/leanderSchulten/git_projekte/vcpkg/buildtrees/wpilib/install-arm64-osx-dbg-err.log</summary>

```
CMake Error at wpimath/cmake_install.cmake:45 (file):
  file INSTALL cannot find
  "/Users/leanderSchulten/git_projekte/vcpkg/buildtrees/wpilib/src/868759e556-81ce2d8562.clean/wpimath/src/main/native/thirdparty/drake/include":
  No such file or directory.
Call Stack (most recent call first):
  cmake_install.cmake:49 (include)
```
</details>

**Additional context**

autoantwort, 2023-07-09:
Code from https://github.com/microsoft/vcpkg/pull/32461

calcmogul, 2023-07-09:
We stopped maintaining the vcpkg package a year ago. See https://github.com/wpilibsuite/allwpilib/pull/4425 for why.

autoantwort, 2023-07-09:
But the error
```
CMake Error at wpimath/cmake_install.cmake:45 (file):
  file INSTALL cannot find
  "/Users/leanderSchulten/git_projekte/vcpkg/buildtrees/wpilib/src/868759e556-81ce2d8562.clean/wpimath/src/main/native/thirdparty/drake/include":
  No such file or directory.
Call Stack (most recent call first):
  cmake_install.cmake:49 (include)
```
should not be vcpkg specific

PR #5427 (2023-07-10): Remove references to Drake
Fixes #5426.
commit fd5699b2 (2023-07-10): Remove references to Drake (#5427)

Fixes #5426.
closed 2023-07-10 by PeterJohnson