wpilibsuite/allwpilib #5428: USE_SYSTEM_FMTLIB not functional

autoantwort, 2023-07-09:
**Describe the bug**
I have specified `USE_SYSTEM_FMTLIB` but it looks like fmtlib is built anyway. 
I get the following error:

```
[43/230] /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/c++  -I/Users/leanderSchulten/git_projekte/vcpkg/buildtrees/wpilib/src/868759e556-81ce2d8562.clean/wpiutil/src/main/native/include -I/Users/leanderSchulten/git_projekte/vcpkg/buildtrees/wpilib/src/868759e556-81ce2d8562.clean/wpiutil/src/main/native/thirdparty/memory/include -I/Users/leanderSchulten/git_projekte/vcpkg/buildtrees/wpilib/src/868759e556-81ce2d8562.clean/wpiutil/src/main/native/thirdparty/json/include -I/Users/leanderSchulten/git_projekte/vcpkg/buildtrees/wpilib/src/868759e556-81ce2d8562.clean/wpiutil/src/main/native/thirdparty/llvm/include -I/Users/leanderSchulten/git_projekte/vcpkg/buildtrees/wpilib/src/868759e556-81ce2d8562.clean/wpiutil/src/main/native/thirdparty/mpack/include -I/Users/leanderSchulten/git_projekte/vcpkg/buildtrees/wpilib/src/868759e556-81ce2d8562.clean/wpiutil/src/main/native/thirdparty/sigslot/include -isystem /Users/leanderSchulten/git_projekte/vcpkg/vcpkg_installed/arm64-osx/include -fPIC -g -arch arm64 -isysroot /Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX13.3.sdk -Wall -pedantic -Wextra -Werror -Wno-unused-parameter -Wno-deprecated-anon-enum-enum-conversion -std=gnu++20 -MD -MT wpiutil/CMakeFiles/wpiutil.dir/src/main/native/thirdparty/fmtlib/src/format.cpp.o -MF wpiutil/CMakeFiles/wpiutil.dir/src/main/native/thirdparty/fmtlib/src/format.cpp.o.d -o wpiutil/CMakeFiles/wpiutil.dir/src/main/native/thirdparty/fmtlib/src/format.cpp.o -c /Users/leanderSchulten/git_projekte/vcpkg/buildtrees/wpilib/src/868759e556-81ce2d8562.clean/wpiutil/src/main/native/thirdparty/fmtlib/src/format.cpp
FAILED: wpiutil/CMakeFiles/wpiutil.dir/src/main/native/thirdparty/fmtlib/src/format.cpp.o 
/Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/c++  -I/Users/leanderSchulten/git_projekte/vcpkg/buildtrees/wpilib/src/868759e556-81ce2d8562.clean/wpiutil/src/main/native/include -I/Users/leanderSchulten/git_projekte/vcpkg/buildtrees/wpilib/src/868759e556-81ce2d8562.clean/wpiutil/src/main/native/thirdparty/memory/include -I/Users/leanderSchulten/git_projekte/vcpkg/buildtrees/wpilib/src/868759e556-81ce2d8562.clean/wpiutil/src/main/native/thirdparty/json/include -I/Users/leanderSchulten/git_projekte/vcpkg/buildtrees/wpilib/src/868759e556-81ce2d8562.clean/wpiutil/src/main/native/thirdparty/llvm/include -I/Users/leanderSchulten/git_projekte/vcpkg/buildtrees/wpilib/src/868759e556-81ce2d8562.clean/wpiutil/src/main/native/thirdparty/mpack/include -I/Users/leanderSchulten/git_projekte/vcpkg/buildtrees/wpilib/src/868759e556-81ce2d8562.clean/wpiutil/src/main/native/thirdparty/sigslot/include -isystem /Users/leanderSchulten/git_projekte/vcpkg/vcpkg_installed/arm64-osx/include -fPIC -g -arch arm64 -isysroot /Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX13.3.sdk -Wall -pedantic -Wextra -Werror -Wno-unused-parameter -Wno-deprecated-anon-enum-enum-conversion -std=gnu++20 -MD -MT wpiutil/CMakeFiles/wpiutil.dir/src/main/native/thirdparty/fmtlib/src/format.cpp.o -MF wpiutil/CMakeFiles/wpiutil.dir/src/main/native/thirdparty/fmtlib/src/format.cpp.o.d -o wpiutil/CMakeFiles/wpiutil.dir/src/main/native/thirdparty/fmtlib/src/format.cpp.o -c /Users/leanderSchulten/git_projekte/vcpkg/buildtrees/wpilib/src/868759e556-81ce2d8562.clean/wpiutil/src/main/native/thirdparty/fmtlib/src/format.cpp
/Users/leanderSchulten/git_projekte/vcpkg/buildtrees/wpilib/src/868759e556-81ce2d8562.clean/wpiutil/src/main/native/thirdparty/fmtlib/src/format.cpp:35:75: error: expected '(' for function-style cast or type construction
                                 basic_format_args<FMT_BUFFER_CONTEXT(char)>,
                                                                      ~~~~^
/Users/leanderSchulten/git_projekte/vcpkg/buildtrees/wpilib/src/868759e556-81ce2d8562.clean/wpiutil/src/main/native/thirdparty/fmtlib/src/format.cpp:34:23: error: explicit instantiation of 'vformat_to' does not refer to a function template, variable template, member function, member class, or static data member
template FMT_API void vformat_to(buffer<char>&, string_view,
                      ^
/Users/leanderSchulten/git_projekte/vcpkg/vcpkg_installed/arm64-osx/include/fmt/format.h:4554:6: note: candidate template ignored: failed template argument deduction
void vformat_to(buffer<Char>& buf, basic_string_view<Char> fmt,
     ^
2 errors generated.
```

**To Reproduce**
Steps to reproduce the behavior:
 Specify `-DUSE_SYSTEM_FMTLIB=ON` and have fmtlib 10.0.0 installed. 

**Expected behavior**
Do not build fmtlib and use the provided one. 

**Additional context**
Happens with the code from the main branch. Used code: https://github.com/microsoft/vcpkg/pull/32461

calcmogul, 2023-07-09:
It works for Arch Linux packages as of 2023.4.3: https://aur.archlinux.org/packages/wpimath

calcmogul, 2023-07-09:
Looks like the wpiutil add_library() call still lists the local fmtlib files.

autoantwort, 2023-07-09:
Yes that is what I also discovered. I have created #5429

closed 2023-07-10 by PeterJohnson