wpilibsuite/allwpilib #5449: Make all pose estimator example projects use Field2d

calcmogul, 2023-07-16:
The differential drive pose estimator example uses Field2d, but not the mecanum or swerve pose estimator examples.

adamgarcia4, 2023-07-26:
Hey Tyler, first time contributing back to the WPILib main repo.

The DifferentialDrivePoseEstimator example defines both an `m_fieldSim` as well as an `m_fieldApproximation`.

`m_fieldSim.setRobotPose(m_drivetrainSimulator.getPose());    m_fieldApproximation.setRobotPose(m_poseEstimator.getEstimatedPosition());`

It seems to me, being that Swerve doesn't have a Drivetrain Sim (please correct me if I'm wrong), we will only need the `m_fieldApproximation` for the SwervePoseEstimator example.

Is this correct?

adamgarcia4, 2023-07-26:
Additionally, running the simulation with my current code changes results in a Pose2d that "jitters" and doesn't respond to inputs.

Would you mind giving me a bit more direction on the matter?

calcmogul, 2023-07-26:
The swerve sim "physics" can just be SwerveDriveKinematics. It would feed `m_fieldSim` as the pose ground truth. The pose estimator would feed `m_fieldApproximation` as a noisy approximation of ground truth.

adamgarcia4, 2023-07-26:
Okay, so please bear with me, as I'm relatively new to FRC Programming.

If I'd like to work out the translational + rotational displacement of the robot wrt the field, I would need to:

Approximate the field displacement by integrating the ChassisSpeed (`new ChassisSpeeds(xSpeed, ySpeed, rot)`) by the `dt` (I guess I can assume as 20ms). Then I guess I could update the m_fieldSim directly from the `drive` method.

I'd get the `dx`, `dy`, and `dTheta` separately, and from there, I can construct the `Pose2d` with which to update the `m_fieldSim`.

Does this approach make sense?

calcmogul, 2023-07-26:
Plug the deltas into Twist2d, then pass that to Pose2d.exp().

The sim update would go in simulationPeriodic(). You can use the diff drive pose estimator as a guide, tho the parts specific to m_drivetrainSimulator will be different.

adamgarcia4, 2023-07-26:
I had done the below, which gets the Field2d to update properly:

```
Transform2d delta = new Transform2d(
	new Translation2d(ySpeed * periodSeconds, xSpeed * periodSeconds),
	new Rotation2d()
);

m_fieldSim.setRobotPose(m_fieldSim.getRobotPose().transformBy(delta));
// m_fieldSim.setRobotPose(m_poseEstimator.getEstimatedPosition());
```

I see that you just mentioned to use a Twist2d to represent the position deltas.
Let me try this.

calcmogul, 2023-07-26:
ySpeed being passed to the x component doesn't seem correct.

adamgarcia4, 2023-07-26:
[image]
Even from reading the docs a month or two ago, I saw that `x` represents translation up/down, why `y` represents strafing. (This is why ChassisSpeeds for differential drive has y=0).

That made me think that I should update the y on the x-component of the Twist.

However, being that you noted this being weird, I've flipped the direction.

calcmogul, 2023-07-26:
The x direction is to the right on the Field2d, and y is up. Heading is measured CCW+ from the x axis.

adamgarcia4, 2023-07-26:
Ahh yes, the ol' Frame of Reference. This makes sense now.
How does my current implementation look for the `m_fieldSim`?

adamgarcia4, 2023-07-26:
Last question before I call it a day and continue tomorrow:
Does the below look correct for updating the Approximation Field2d?

`// m_fieldApproximation.setRobotPose(m_poseEstimator.getEstimatedPosition());`

calcmogul, 2023-07-26:
The m_fieldSim update looks fine at a glance.

The m_fieldApproximation also looks correct.

adamgarcia4, 2023-07-26:
Great, I'll continue on it tomorrow.
This back/forth helped me quite substantially. Thanks!

adamgarcia4, 2023-07-28:
For some reason, the `SwerveDrivePoseEstimator` isn't updating properly.
No code has been changed from this point of view.
https://github.com/wpilibsuite/allwpilib/assets/7979621/f6c58c77-9b3d-47dc-adcf-e1846606913e

adamgarcia4, 2023-07-28:
I've investigated, and I have arrived at the following conclusion:

Nowhere in the Swerve Drive Pose Estimator example are the Encoders being simulated. This is why the Field2d Field approximation isn't moving.

The differentialdriveposeestimator sets both the encoder sims `setDistance` and `setRate` according to the output of the drivetrain simulator.

As we don't (yet!) have a swerveSimulator, we are going to have to hook into the `setDesiredState` of the SwerveModule.

These are my current findings. I'm currently struggling with getting these updating properly.

The `SwerveModuleState` consists of a wheel SPEED, and an angular POSITION.

I can track the total distance the wheel has traveled by `speedOfWheel * dt`, and get the rough angular rate by tracking the last angle and `rateDiff/dt`. (essentially numerically integrating distance to velocity, then tracking the derivative to get angularRate.

Does this sound like a reasonable approach?

calcmogul, 2023-07-28:
A function that's the inverse of SwerveDriveKinematics.toTwist2d() would be able to turn a chassis twist from Pose2d.log() into wheel position deltas.

adamgarcia4, 2023-07-30:
I'm making progress, and rediscovering the mathematics from my undergrad in the process!
Btw, I'm learning a lot from this Robotics course from Northwestern:
https://youtube.com/playlist?list=PLggLP4f-rq02vX0OQQ5vrCxbJrzamYDfx

Update: I have added the relevant updates to the Swerve Drive Encoder Sims as shown in this commit:
https://github.com/wpilibsuite/allwpilib/pull/5471/commits/28a1a629a1193ae8af38c36b3c0ecf389934c869

You can see the resulting updates in the Simulation. I know there must be a factor off in the `setDistance` updates, as the encoder expects values in encoder ticks. Maybe you can provide assistance in that area.

https://github.com/wpilibsuite/allwpilib/assets/7979621/f16a1f86-4183-4f2a-baf9-56d6c87730b4

adamgarcia4, 2023-07-30:
With regards to your earlier statement about putting this logic in the SwerveDriveKinematics, I'm struggling a bit.

Let me clarify my current understanding.

From your Controls System book Sec. 10.2, We can take a `pose2d`, which is in SE(2), and apply a `Twist2d` via an expoential to arrive at the new `pose2d`. As defined in the `pose2d` class, you also have a `log()` function which backs out the relevant `Twist2d`.

This is the relationship between taking the exponent and reversing that via the log, as it applies to coordinate systems. (as I understand it).

As it applies to the `SwerveDriveKinematics`, we have a series of functions which leverage both the inverseKinematics and forwardKinematics.

We have defined a `toTwist2d`, which can take in `moduleDeltas` and back out the relevant `Twist2d`. These moduleDeltas come from encoder/gyro readings.

I believe what you proposed is the opposite of this, where a `Twist2d` is the input, and the relevant moduleDeltas are the output. These can then finally be used to set the relevant Swerve Module Encoder/Gyro Sims.

I guess my next step is to actually follow the derivation of the kinematics for Swerve that you've used to understand the problem space more. With that information, I could attempt to write the `SwerveModulePosition[] toModuleDeltas(Twist2d twist)` method. Am I on the right track here?

adamgarcia4, 2023-07-30:
Oh, I just looked at the derivation. Very easy to understand the Matrix `M` is to compute the Swerve Module State from the `ChasisSpeed` (inverse kinematics) and the `M+` for getting the `ChasisSpeed`  from the Swerve Module State (forward inverse).