wpilibsuite/allwpilib #5460: Symbols missing from 2023.4.3 linuxathena version of wpilibNewCommands shared library file

theVerySharpFlat, 2023-07-19:
**Describe the bug**
A clear and concise description of what the bug is.

In short, the symbols `frc2::Command::GetRequirements() const` and `frc2::Command::InitSendable(wpi::SendableBuilder&)' are missing from the linuxathena 2023.4.3 release version of the wpilibNewCommands library, causing a linker error in my code.

This bug was detected in my vendor dependency <https://github.com/rambunction4330/librmb/tree/linker-error-fix> (on the branch linked).

**To Reproduce**
Steps to reproduce the behavior:

Build the above project (on the above branch) as follows
1. `./gradlew installRoboRioToolchain`
2. `./gradlew build`

Build fails with error:
```
/home/gitpod/.gradle/toolchains/frc/2023/roborio/bin/../arm-nilrt-linux-gnueabi/sysroot/usr/lib/gcc/arm-nilrt-linux-gnueabi/12/../../../../../../../arm-nilrt-linux-gnueabi/bin/ld: /workspace/librmb/build/libs/libRmb/shared/linuxathena/release/libLibRmb.so: undefined reference to `frc2::Command::GetRequirements() const'
/home/gitpod/.gradle/toolchains/frc/2023/roborio/bin/../arm-nilrt-linux-gnueabi/sysroot/usr/lib/gcc/arm-nilrt-linux-gnueabi/12/../../../../../../../arm-nilrt-linux-gnueabi/bin/ld: /workspace/librmb/build/libs/libRmb/shared/linuxathena/release/libLibRmb.so: undefined reference to `frc2::Command::InitSendable(wpi::SendableBuilder&)'
collect2: error: ld returned 1 exit status
```

My project consists of a native library vendor dependency in the root directory and an executable in project `:testbench` that links to the root library using `lib project ":" library "LibRmb"`

Building the library works without a hitch as `ld` is able to find the symbols just fine in the `2023.4.3-122-g335e7dd` version of the .so. However, when building the executable the build system (gradle, ew) tries to link with the `2023.4.3` stable release version of the .so which does not contain these symbols.

To test my hypothesis that the symbols were missing, I ran the following:
```shell
$ readelf -WCs /workspace/.gradle/caches/transforms-3/45d8e6036d6c036e6a2524b1b063fb8a/transformed/wpilibNewCommands-cpp-2023.4.3-122-g335e7dd-linuxathenadebug/linux/athena/shared/libwpilibNewCommandsd.so |\
> grep frc2::Command::GetRequirements
  1818: 0006baa0    28 FUNC    GLOBAL DEFAULT   12 frc2::Command::GetRequirements() const
  5046: 0006baa0    28 FUNC    GLOBAL DEFAULT   12 frc2::Command::GetRequirements() const
```

However, the same with the stable release yields nothing:
```shell
$ readelf -WCs /workspace/.gradle/caches/transforms-3/8c2f9fe4125490231b7f7fb220403aed/transformed/wpilibNewCommands-cpp-2023.4.3-linuxathenadebug/linux/athena/shared/libwpilibNewCommandsd.so |\
> grep frc2::Command::GetRequirements
$
```

It seems that this is only the case with linuxathena as I've tested desktop builds on both darwin-arm64/clang and linux-x86-64/gcc and they work.

 - Link to code:
 <https://github.com/rambunction4330/librmb/tree/linker-error-fix>
* Vendor dependency library is the root project
* Testbench executable is under `:/testbench`

**Expected behavior**
I expected readelf to show that `libwpilibNewCommands-2023.4.3.so` contains the symbols mentioned above and that my build succeeds without linker errors.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Desktop (please complete the following information):**
 - OS: darwin-arm64/apple-clang, linux-x86-64/gcc
 - Project Information: Not using VSCode
 
RoboRIO toolchain, gcc 12

Library Dependencies:
```groovy
nativeUtils {
    wpi {
        configureDependencies {
            wpiVersion = "2023.+"
            niLibVersion = "2023.+"
            opencvVersion = "4.6.0-3"
            wpimathVersion = "2023.+"
        }
    }
}
```

Testbench Executable WPILib version 2023.4.3 (I think it pulls the latest stable off of the wpi maven repo)

**Additional context**
Add any other context about the problem here.

I understand that the way I'm using GradleRIO is unorthodox. I hope my problem isn't self-created.

sciencewhiz, 2023-07-19:
The symbols aren't "missing" in 2023.4.3, They were added in https://github.com/wpilibsuite/allwpilib/pull/5392 a few dayss ago.

See: 
https://github.com/rambunction4330/librmb/blob/e046814019e4fc1609a07b019c22bc8d0061447d/build.gradle#L16-L20

You need to call `./gradlew build -PreleaseMode`, so that it builds with the 2023 release, rather then current main, which has the change.

When I build your code on Windows, I don't get linker errors, but do get compile errors related to some of your SparkMax code.

theVerySharpFlat, 2023-07-19:
> The symbols aren't "missing" in 2023.4.3, They were added in #5392 a few days ago and thus shouldn't exist. The problem appears to be mixing of development and release builds.

Wow. I didn't even consider that. If only I had thought a little bit more about the differing version numbers!

> You need to call `./gradlew build -PreleaseMode`, so that it builds with the 2023 release, rather then current main, which has the change.

Is that a GradleRIO flag? This fixed everything!

> When I build your code on Windows, I don't get linker errors, but do get compile errors related to some of your SparkMax code.

I'll be sure to check this out on my windows machine. Thanks for all your help!!!

closed 2023-07-19 by theVerySharpFlat

theVerySharpFlat, 2023-07-19:
Ok `-PreleaseMode` is part of the vendor template.