wpilibsuite/allwpilib #5467: Command unscheduling order is inconsistent

Starlight220, 2023-07-24:
>>> The command scheduler actually contains three separate implementations of “call Command.end() and unschedule”. For two of them, the command is unscheduled after calling end, but in the cancel case it is unscheduled first. A command’s end method can therefore detect whether an interruption is from cancel or from the robot becoming disabled by calling isScheduled. This is the only case I know of where a life cycle method is called on an unscheduled command.
Edit: These remarks apply only to the Java implementation. The C++ implementation of CommandScheduler behaves slightly differently. There are only two implementations of “call end and unschedule”, and it therefore calls Command.end on an unscheduled command in both interruption cases.
I assumed that the CommandScheduler guaranteed that life cycle methods would only be invoked on scheduled commands, but it turns out that I was wrong.

Source: [Chief Delphi](https://www.chiefdelphi.com/t/commandbase-subsystembase-depreciation/438975/181?u=starlight220)

Starlight220, 2023-07-24:
Seems like #4261 is somewhat to blame, meaning its cause issue needs to be taken into consideration. The question is how to make this robust.

Gold856, 2023-07-24:
Another inconsistency: sometimes, the command is unscheduled and the requirements are dropped, then the actions (finished or interrupted) are run (this applies to cancel), but other times, the actions are run, then the command is unscheduled and requirements are dropped (this applies to finished commands). Which order do we want to have?

PR #5470 (2023-10-10): Make command scheduling order consistent
Fixes #5467.
`end` is now called first along with interrupt actions on both languages. Sets are used to defer cancels, ensuring infinite loops aren't possible.
closed 2023-10-10 by PeterJohnson