wpilibsuite/allwpilib #5514: Installation on windows fails

autoantwort, 2023-08-06:
**Describe the bug**
When I try to install the project on windows it fails to install

**To Reproduce**
Steps to reproduce the behavior:
1. Checkout https://github.com/microsoft/vcpkg/pull/32461
2. `./vcpkg install wpilib`

Package: wpilib:x64-windows -> 2023-08-05

**Host Environment**

- Host: x64-windows
- Compiler: MSVC 19.36.32537.0
-    vcpkg-tool version: 2999-12-31-unknownhash
    vcpkg-scripts version: 748037b39 2023-08-05 (16 hours ago)

**To Reproduce**

`vcpkg install --enforce-port-checks --allow-unsupported`

**Failure logs**

<details><summary>Failure logs</summary>

```
[1/2] "C:/Users/LeanderSchulten/git_projects/vcpkg/downloads/tools/cmake-3.25.1-windows/cmake-3.25.1-windows-i386/bin/cmake.exe" -E chdir ".." "C:/Users/LeanderSchulten/git_projects/vcpkg/downloads/tools/cmake-3.25.1-windows/cmake-3.25.1-windows-i386/bin/cmake.exe" "C:/Users/LeanderSchulten/git_projects/vcpkg/buildtrees/wpilib/src/4f547bd2c6-e358961f33.clean" "-G" "Ninja" "-DCMAKE_BUILD_TYPE=Release" "-DCMAKE_INSTALL_PREFIX=C:/Users/LeanderSchulten/git_projects/vcpkg/packages/wpilib_x64-windows" "-DFETCHCONTENT_FULLY_DISCONNECTED=ON" "-DWITH_CSCORE=OFF" "-DWITH_SIMULATION_MODULES=OFF" "-DWITH_WPILIB=OFF" "-DPKG_CONFIG_EXECUTABLE=C:/Users/LeanderSchulten/git_projects/vcpkg/downloads/tools/msys2/6f3fa1a12ef85a6f/mingw32/bin/pkg-config.exe" "-DWITH_JAVA=OFF" "-DWITH_EXAMPLES=OFF" "-DWITH_TESTS=OFF" "-DWITH_GUI=OFF" "-DWITH_SIMULATION_MODULES=OFF" "-DUSE_SYSTEM_FMTLIB=ON" "-DUSE_SYSTEM_LIBUV=ON" "-DUSE_SYSTEM_EIGEN=ON" "-DCMAKE_MAKE_PROGRAM=C:/Users/LeanderSchulten/git_projects/vcpkg/downloads/tools/ninja/1.10.2-windows/ninja.exe" "-DBUILD_SHARED_LIBS=ON" "-DVCPKG_CHAINLOAD_TOOLCHAIN_FILE=C:/Users/LeanderSchulten/git_projects/vcpkg/scripts/toolchains/windows.cmake" "-DVCPKG_TARGET_TRIPLET=x64-windows" "-DVCPKG_SET_CHARSET_FLAG=ON" "-DVCPKG_PLATFORM_TOOLSET=v143" "-DCMAKE_EXPORT_NO_PACKAGE_REGISTRY=ON" "-DCMAKE_FIND_PACKAGE_NO_PACKAGE_REGISTRY=ON" "-DCMAKE_FIND_PACKAGE_NO_SYSTEM_PACKAGE_REGISTRY=ON" "-DCMAKE_INSTALL_SYSTEM_RUNTIME_LIBS_SKIP=TRUE" "-DCMAKE_VERBOSE_MAKEFILE=ON" "-DVCPKG_APPLOCAL_DEPS=OFF" "-DCMAKE_TOOLCHAIN_FILE=C:/Users/LeanderSchulten/git_projects/vcpkg/scripts/buildsystems/vcpkg.cmake" "-DCMAKE_ERROR_ON_ABSOLUTE_INSTALL_DESTINATION=ON" "-DVCPKG_CXX_FLAGS=" "-DVCPKG_CXX_FLAGS_RELEASE=" "-DVCPKG_CXX_FLAGS_DEBUG=" "-DVCPKG_C_FLAGS=" "-DVCPKG_C_FLAGS_RELEASE=" "-DVCPKG_C_FLAGS_DEBUG=" "-DVCPKG_CRT_LINKAGE=dynamic" "-DVCPKG_LINKER_FLAGS=" "-DVCPKG_LINKER_FLAGS_RELEASE=" "-DVCPKG_LINKER_FLAGS_DEBUG=" "-DVCPKG_TARGET_ARCHITECTURE=x64" "-DCMAKE_INSTALL_LIBDIR:STRING=lib" "-DCMAKE_INSTALL_BINDIR:STRING=bin" "-D_VCPKG_ROOT_DIR=C:/Users/LeanderSchulten/git_projects/vcpkg" "-D_VCPKG_INSTALLED_DIR=C:/Users/LeanderSchulten/git_projects/vcpkg/vcpkg_installed" "-DVCPKG_MANIFEST_INSTALL=OFF"
FAILED: ../CMakeCache.txt 
"C:/Users/LeanderSchulten/git_projects/vcpkg/downloads/tools/cmake-3.25.1-windows/cmake-3.25.1-windows-i386/bin/cmake.exe" -E chdir ".." "C:/Users/LeanderSchulten/git_projects/vcpkg/downloads/tools/cmake-3.25.1-windows/cmake-3.25.1-windows-i386/bin/cmake.exe" "C:/Users/LeanderSchulten/git_projects/vcpkg/buildtrees/wpilib/src/4f547bd2c6-e358961f33.clean" "-G" "Ninja" "-DCMAKE_BUILD_TYPE=Release" "-DCMAKE_INSTALL_PREFIX=C:/Users/LeanderSchulten/git_projects/vcpkg/packages/wpilib_x64-windows" "-DFETCHCONTENT_FULLY_DISCONNECTED=ON" "-DWITH_CSCORE=OFF" "-DWITH_SIMULATION_MODULES=OFF" "-DWITH_WPILIB=OFF" "-DPKG_CONFIG_EXECUTABLE=C:/Users/LeanderSchulten/git_projects/vcpkg/downloads/tools/msys2/6f3fa1a12ef85a6f/mingw32/bin/pkg-config.exe" "-DWITH_JAVA=OFF" "-DWITH_EXAMPLES=OFF" "-DWITH_TESTS=OFF" "-DWITH_GUI=OFF" "-DWITH_SIMULATION_MODULES=OFF" "-DUSE_SYSTEM_FMTLIB=ON" "-DUSE_SYSTEM_LIBUV=ON" "-DUSE_SYSTEM_EIGEN=ON" "-DCMAKE_MAKE_PROGRAM=C:/Users/LeanderSchulten/git_projects/vcpkg/downloads/tools/ninja/1.10.2-windows/ninja.exe" "-DBUILD_SHARED_LIBS=ON" "-DVCPKG_CHAINLOAD_TOOLCHAIN_FILE=C:/Users/LeanderSchulten/git_projects/vcpkg/scripts/toolchains/windows.cmake" "-DVCPKG_TARGET_TRIPLET=x64-windows" "-DVCPKG_SET_CHARSET_FLAG=ON" "-DVCPKG_PLATFORM_TOOLSET=v143" "-DCMAKE_EXPORT_NO_PACKAGE_REGISTRY=ON" "-DCMAKE_FIND_PACKAGE_NO_PACKAGE_REGISTRY=ON" "-DCMAKE_FIND_PACKAGE_NO_SYSTEM_PACKAGE_REGISTRY=ON" "-DCMAKE_INSTALL_SYSTEM_RUNTIME_LIBS_SKIP=TRUE" "-DCMAKE_VERBOSE_MAKEFILE=ON" "-DVCPKG_APPLOCAL_DEPS=OFF" "-DCMAKE_TOOLCHAIN_FILE=C:/Users/LeanderSchulten/git_projects/vcpkg/scripts/buildsystems/vcpkg.cmake" "-DCMAKE_ERROR_ON_ABSOLUTE_INSTALL_DESTINATION=ON" "-DVCPKG_CXX_FLAGS=" "-DVCPKG_CXX_FLAGS_RELEASE=" "-DVCPKG_CXX_FLAGS_DEBUG=" "-DVCPKG_C_FLAGS=" "-DVCPKG_C_FLAGS_RELEASE=" "-DVCPKG_C_FLAGS_DEBUG=" "-DVCPKG_CRT_LINKAGE=dynamic" "-DVCPKG_LINKER_FLAGS=" "-DVCPKG_LINKER_FLAGS_RELEASE=" "-DVCPKG_LINKER_FLAGS_DEBUG=" "-DVCPKG_TARGET_ARCHITECTURE=x64" "-DCMAKE_INSTALL_LIBDIR:STRING=lib" "-DCMAKE_INSTALL_BINDIR:STRING=bin" "-D_VCPKG_ROOT_DIR=C:/Users/LeanderSchulten/git_projects/vcpkg" "-D_VCPKG_INSTALLED_DIR=C:/Users/LeanderSchulten/git_projects/vcpkg/vcpkg_installed" "-DVCPKG_MANIFEST_INSTALL=OFF"
-- The C compiler identification is MSVC 19.36.32537.0
-- The CXX compiler identification is MSVC 19.36.32537.0
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: C:/Program Files/Microsoft Visual Studio/2022/Community/VC/Tools/MSVC/14.36.32532/bin/Hostx64/x64/cl.exe - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: C:/Program Files/Microsoft Visual Studio/2022/Community/VC/Tools/MSVC/14.36.32532/bin/Hostx64/x64/cl.exe - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Platform version: 10.0.18362.0
-- Could NOT find LIBSSH (missing: LIBSSH_INCLUDE_DIRS LIBSSH_LIBRARIES) (Required is at least version "0.7.1")
-- Performing Test CMAKE_HAVE_LIBC_PTHREAD
-- Performing Test CMAKE_HAVE_LIBC_PTHREAD - Failed
-- Looking for pthread_create in pthreads
-- Looking for pthread_create in pthreads - not found
-- Looking for pthread_create in pthread
-- Looking for pthread_create in pthread - not found
-- Found Threads: TRUE  
CMake Error at wpiutil/CMakeLists.txt:187 (install):
  install FILES given no DESTINATION!

CMake Error at wpiutil/CMakeLists.txt:188 (install):
  install EXPORT given no DESTINATION!

CMake Error at wpinet/CMakeLists.txt:187 (install):
  install FILES given no DESTINATION!

CMake Error at wpinet/CMakeLists.txt:188 (install):
  install EXPORT given no DESTINATION!

CMake Error at ntcore/CMakeLists.txt:47 (install):
  install FILES given no DESTINATION!

CMake Error at ntcore/CMakeLists.txt:48 (install):
  install EXPORT given no DESTINATION!

CMake Error at wpimath/CMakeLists.txt:142 (install):
  install FILES given no DESTINATION!
[... 739 lines cut ...]
//ADVANCED property for variable: CPACK_BINARY_NSIS
CPACK_BINARY_NSIS-ADVANCED:INTERNAL=1
//ADVANCED property for variable: CPACK_BINARY_NUGET
CPACK_BINARY_NUGET-ADVANCED:INTERNAL=1
//ADVANCED property for variable: CPACK_BINARY_WIX
CPACK_BINARY_WIX-ADVANCED:INTERNAL=1
//ADVANCED property for variable: CPACK_BINARY_ZIP
CPACK_BINARY_ZIP-ADVANCED:INTERNAL=1
//ADVANCED property for variable: CPACK_SOURCE_7Z
CPACK_SOURCE_7Z-ADVANCED:INTERNAL=1
//ADVANCED property for variable: CPACK_SOURCE_ZIP
CPACK_SOURCE_ZIP-ADVANCED:INTERNAL=1
//Details about finding Threads
FIND_PACKAGE_MESSAGE_DETAILS_Threads:INTERNAL=[TRUE][v()]
//ADVANCED property for variable: VCPKG_VERBOSE
VCPKG_VERBOSE-ADVANCED:INTERNAL=1
//Making sure VCPKG_MANIFEST_MODE doesn't change
Z_VCPKG_CHECK_MANIFEST_MODE:INTERNAL=ON
//Vcpkg root directory
Z_VCPKG_ROOT_DIR:INTERNAL=C:/Users/LeanderSchulten/git_projects/vcpkg

```
</details>

The problem seems to be that on MSVC `WITH_FLAT_INSTALL` is set to ON but `wpilib_dest` to the empty string. As a result the `DESTINATION` parameter in the install command is an empty string.

PeterJohnson, 2023-08-06:
Probably broken by #4492?  We no longer test build vcpkg in CI due to disk space limitations and oddities occurring in vcpkg dependency install (removed in #4425) so it’s not surprising this broke.

autoantwort, 2023-08-06:
Probably. `WITH_FLAT_INSTALL` seems to be broken on all platforms, but `WITH_FLAT_INSTALL` is force enables on MSVC. 

With the main branch on macOS
<details><summary>Log:</summary>

```
➜  allwpilib git:(main) cmake -S . -B build -GNinja -DWITH_FLAT_INSTALL=ON                                           
CMake Deprecation Warning at CMakeLists.txt:14 (cmake_minimum_required):
  Compatibility with CMake < 3.5 will be removed from a future version of
  CMake.

  Update the VERSION argument <min> value or use a ...<max> suffix to tell
  CMake that the project does not need compatibility with older versions.

-- The C compiler identification is AppleClang 14.0.3.14030022
-- The CXX compiler identification is AppleClang 14.0.3.14030022
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/cc - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/c++ - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Platform version: 
-- Found LIBSSH: /opt/homebrew/include (found suitable version "0.10.5", minimum required is "0.7.1") 
-- Found Python: /opt/homebrew/Frameworks/Python.framework/Versions/3.11/bin/python3.11 (found version "3.11.4") found components: Interpreter 
-- Performing Test CMAKE_HAVE_LIBC_PTHREAD
-- Performing Test CMAKE_HAVE_LIBC_PTHREAD - Success
-- Found Threads: TRUE  
-- Found Java: /Library/Java/JavaVirtualMachines/adoptopenjdk-8.jdk/Contents/Home/bin/java (found version "1.8.0.292") 
-- Found JNI: /Library/Java/JavaVirtualMachines/adoptopenjdk-8.jdk/Contents/Home/include  found components: AWT JVM 
-- Downloading Jackson jarfiles...
-- All files downloaded.
CMake Error at wpiutil/CMakeLists.txt:187 (install):
  install FILES given no DESTINATION!

CMake Error at wpiutil/CMakeLists.txt:188 (install):
  install EXPORT given no DESTINATION!

CMake Error at wpinet/CMakeLists.txt:187 (install):
  install FILES given no DESTINATION!

CMake Error at wpinet/CMakeLists.txt:188 (install):
  install EXPORT given no DESTINATION!

CMake Error at ntcore/CMakeLists.txt:47 (install):
  install FILES given no DESTINATION!

CMake Error at ntcore/CMakeLists.txt:48 (install):
  install EXPORT given no DESTINATION!

-- Downloading EJML jarfiles...
-- All files downloaded.
CMake Error at wpimath/CMakeLists.txt:142 (install):
  install FILES given no DESTINATION!

CMake Error at wpimath/CMakeLists.txt:143 (install):
  install EXPORT given no DESTINATION!

CMake Warning (dev) at /opt/homebrew/Cellar/cmake/3.27.1/share/cmake/Modules/FetchContent.cmake:1316 (message):
  The DOWNLOAD_EXTRACT_TIMESTAMP option was not given and policy CMP0135 is
  not set.  The policy's OLD behavior will be used.  When using a URL
  download, the timestamps of extracted files should preferably be that of
  the time of extraction, otherwise code that depends on the extracted
  contents might not be rebuilt if the URL changes.  The OLD behavior
  preserves the timestamps from the archive instead, but this is usually not
  what you want.  Update your project to the NEW behavior or specify the
  DOWNLOAD_EXTRACT_TIMESTAMP option with a value of true to avoid this
  robustness issue.
Call Stack (most recent call first):
  imgui/CMakeLists.txt:23 (FetchContent_Declare)
This warning is for project developers.  Use -Wno-dev to suppress it.

-- Found Doxygen: /Applications/Doxygen.app/Contents/Resources/doxygen (found version "1.8.14") found components: doxygen 
-- Including Cocoa support
-- Found Python: /opt/homebrew/Frameworks/Python.framework/Versions/3.11/bin/python3.11 (found version "3.11.4") found components: Interpreter Development Development.Module Development.Embed 
Downloading /Users/leanderSchulten/git_projekte/allwpilib/build/_deps/gl3w-build/include/GL/glcorearb.h...
Downloading /Users/leanderSchulten/git_projekte/allwpilib/build/_deps/gl3w-build/include/KHR/khrplatform.h...
Parsing glcorearb.h header...
Generating /Users/leanderSchulten/git_projekte/allwpilib/build/_deps/gl3w-build/include/GL/gl3w.h...
Generating /Users/leanderSchulten/git_projekte/allwpilib/build/_deps/gl3w-build/src/gl3w.c...
CMake Error at hal/CMakeLists.txt:69 (install):
  install FILES given no DESTINATION!

CMake Error at hal/CMakeLists.txt:70 (install):
  install EXPORT given no DESTINATION!

CMake Error at cscore/CMakeLists.txt:8 (find_package):
  By not providing "FindOpenCV.cmake" in CMAKE_MODULE_PATH this project has
  asked CMake to find a package configuration file provided by "OpenCV", but
  CMake did not find one.

  Could not find a package configuration file provided by "OpenCV" with any
  of the following names:

    OpenCVConfig.cmake
    opencv-config.cmake

  Add the installation prefix of "OpenCV" to CMAKE_PREFIX_PATH or set
  "OpenCV_DIR" to a directory containing one of the above files.  If "OpenCV"
  provides a separate development package or SDK, be sure it has been
  installed.

-- Configuring incomplete, errors occurred!
```

</details>

closed 2023-08-07 by PeterJohnson