wpilibsuite/allwpilib #5518: TelescopingArm physics sim

calcmogul, 2023-08-07:
**Is your feature request related to a problem? Please describe.**
Many teams had telescoping arms in the 2023 season, but we don't have a physics sim class for that.

**Describe the solution you'd like**
We should derive a coupled model using Lagrangian mechanics (Newtonian mechanics may be annoying for this), then make a wrapper class for it.

**Describe alternatives you've considered**
Letting users modify the arm length in SingleJointedArmSim is an option, but it doesn't capture the centripetal acceleration on the arm from the arm rotating.

srimanachanta, 2023-08-18:
Duplicate #5289

calcmogul, 2023-08-18:
#5289 is a separate issue about a "spring and winch mechanism sim". A telescoping arm sim class wouldn't have anything related to springs or winches.