wpilibsuite/allwpilib #5519: DataLog.cpp fails to build on windows

autoantwort, 2023-08-07:
**Describe the bug**

```
D:\a\allwpilib\allwpilib\wpiutil\src\main\native\cpp\DataLog.cpp(201,1): error C2760: syntax error: unexpected token ',', expected ')' [D:\a\allwpilib\allwpilib\build\wpiutil\wpiutil.vcxproj]
D:\a\allwpilib\allwpilib\wpiutil\src\main\native\cpp\DataLog.cpp(201,1): error C2760: syntax error: unexpected token ')', expected 'expression' [D:\a\allwpilib\allwpilib\build\wpiutil\wpiutil.vcxproj]
D:\a\allwpilib\allwpilib\wpiutil\src\main\native\cpp\DataLog.cpp(201,1): error C2187: syntax error: ')' was unexpected here [D:\a\allwpilib\allwpilib\build\wpiutil\wpiutil.vcxproj]
D:\a\allwpilib\allwpilib\wpiutil\src\main\native\cpp\DataLog.cpp(202,1): error C2109: subscript requires array or pointer type [D:\a\allwpilib\allwpilib\build\wpiutil\wpiutil.vcxproj]
  gmock_main.cc
D:\a\allwpilib\allwpilib\wpiutil\src\main\native\cpp\DataLog.cpp(201,1): error C2059: syntax error: ')' [D:\a\allwpilib\allwpilib\build\wpiutil\wpiutil.vcxproj]
D:\a\allwpilib\allwpilib\wpiutil\src\main\native\cpp\DataLog.cpp(201,1): error C2065: 'char_type': undeclared identifier [D:\a\allwpilib\allwpilib\build\wpiutil\wpiutil.vcxproj]
D:\a\allwpilib\allwpilib\wpiutil\src\main\native\cpp\DataLog.cpp(201,1): error C2923: 'fmt::v9::basic_string_view': 'char_type' is not a valid template type argument for parameter 'Char' [D:\a\allwpilib\allwpilib\build\wpiutil\wpiutil.vcxproj]
D:\a\allwpilib\allwpilib\wpiutil\src\main\native\cpp\DataLog.cpp(201): message : see declaration of 'char_type' [D:\a\allwpilib\allwpilib\build\wpiutil\wpiutil.vcxproj]
D:\a\allwpilib\allwpilib\wpiutil\src\main\native\cpp\DataLog.cpp(202,1): error C2270: 'operator fmt::v9::basic_string_view': modifiers not allowed on nonmember functions [D:\a\allwpilib\allwpilib\build\wpiutil\wpiutil.vcxproj]
D:\a\allwpilib\allwpilib\wpiutil\src\main\native\cpp\DataLog.cpp(202,1): error C2955: 'fmt::v9::basic_string_view': use of class template requires template argument list [D:\a\allwpilib\allwpilib\build\wpiutil\wpiutil.vcxproj]
D:\a\allwpilib\allwpilib\wpiutil\src\main\native\thirdparty\fmtlib\include\fmt\core.h(430): message : see declaration of 'fmt::v9::basic_string_view' [D:\a\allwpilib\allwpilib\build\wpiutil\wpiutil.vcxproj]
D:\a\allwpilib\allwpilib\wpiutil\src\main\native\cpp\DataLog.cpp(201,1): error C2601: 'operator fmt::v9::basic_string_view': local function definitions are illegal [D:\a\allwpilib\allwpilib\build\wpiutil\wpiutil.vcxproj]
D:\a\allwpilib\allwpilib\wpiutil\src\main\native\cpp\DataLog.cpp(202): message : this line contains a '{' which has not yet been matched [D:\a\allwpilib\allwpilib\build\wpiutil\wpiutil.vcxproj]
D:\a\allwpilib\allwpilib\wpiutil\src\main\native\cpp\DataLog.cpp(201,1): error C3861: 'FMT_COMPILE_STRING': identifier not found [D:\a\allwpilib\allwpilib\build\wpiutil\wpiutil.vcxproj]
D:\a\allwpilib\allwpilib\wpiutil\src\main\native\cpp\DataLog.cpp(202,1): error C2059: syntax error: ')' [D:\a\allwpilib\allwpilib\build\wpiutil\wpiutil.vcxproj]
D:\a\allwpilib\allwpilib\wpiutil\src\main\native\cpp\DataLog.cpp(202,32): error C2059: syntax error: ';' [D:\a\allwpilib\allwpilib\build\wpiutil\wpiutil.vcxproj]
D:\a\allwpilib\allwpilib\wpiutil\src\main\native\cpp\DataLog.cpp(201,1): error C2143: syntax error: missing ')' before ',' [D:\a\allwpilib\allwpilib\build\wpiutil\wpiutil.vcxproj]
D:\a\allwpilib\allwpilib\wpiutil\src\main\native\cpp\DataLog.cpp(201,1): error C2064: term does not evaluate to a function taking 0 arguments [D:\a\allwpilib\allwpilib\build\wpiutil\wpiutil.vcxproj]
D:\a\allwpilib\allwpilib\wpiutil\src\main\native\cpp\DataLog.cpp(201,1): error C3615: constexpr function 'operator fmt::v9::basic_string_view' cannot result in a constant expression [D:\a\allwpilib\allwpilib\build\wpiutil\wpiutil.vcxproj]
D:\a\allwpilib\allwpilib\wpiutil\src\main\native\cpp\DataLog.cpp(202): message : failure was caused by control reaching the end of a constexpr function [D:\a\allwpilib\allwpilib\build\wpiutil\wpiutil.vcxproj]
```

**To Reproduce**
Steps to reproduce the behavior:
See windows pipeline in https://github.com/wpilibsuite/allwpilib/pull/5516. For example https://github.com/wpilibsuite/allwpilib/actions/runs/5783985198/job/15673865372?pr=5516

**Additional context**
Is `DataLog.cpp` also build with the gradle build system? I have not found build instruction in gradle for `DataLog.cpp` respectively for `main/native/cpp` in `wpiutil`.

ThadHouse, 2023-08-07:
Cmake probably needs to be bumped to c++20 I’m guessing. 

Everything in that src folder is built with gradle, it’s a wildcard.

calcmogul, 2023-08-07:
```
wpiutil/CMakeLists.txt
108:target_compile_features(wpiutil PUBLIC cxx_std_20)
```

ThadHouse, 2023-08-08:
Try adding `/Zc:preprocessor` to https://github.com/wpilibsuite/allwpilib/blob/6b896a38dcf774d0eee241b2d8696cae8c41681b/wpiutil/CMakeLists.txt#L110

closed 2023-08-17 by PeterJohnson