wpilibsuite/allwpilib #5522: DualSense Support for PS5 Controller

a1cd, 2023-08-09:
**Is your feature request related to a problem? Please describe.**
I really would like to be able to interface with the PS5 DualSense controller's advantages, for example having trigger force feedback. It would be nice to be able to give the driver feedback while pulling the trigger, for example while pushing another robot. 

This could be expanded to the touchpad on the DUALSHOCK series and on the PS5 controllers. If this is possible, API access to the light-bar (setting the light-bar color) could possibly be an extension to this idea. It could be useful for identifying a controller belonging to a drive team role. (if these are truly separate issues and would require separate work I can create another issue for each)

**Describe the solution you'd like**
I would like to have API access to dual sense controller features, ideally in a similar way to the API for controller rumble. I would be able to set the resistance on the left trigger to 100% or the right trigger to the amount of current on a motor. 

**Describe alternatives you've considered**
I would be ok with downloading a package on the drive station and a package on the robot to enable extra communication as well, allowing for that extra data. This may be the only solution if it would require changes to the driver station's software. 
One way this could be done is emulating two controllers for one ps5 controller, one being the dual sense features (low rumble = left trigger, high rumble = right trigger, maybe include the gyroscope, maybe the touchpad as well) and have the robot combine them into one controller. 

**Additional context**
Here is related Dualsense and PS5 controller content
[**Reddit** – DualSense Haptics, LEDs and more (HID Output Report)](https://web.archive.org/web/20201228212942/https://www.reddit.com/r/gamedev/comments/jumvi5/dualsense_haptics_leds_and_more_hid_output_report/) (web archive because it was since deleted)
There is not as much in java about it but if you search up the GitHub "dual sense" topic there is plenty in c#, c++, c, and python.

rzblue, 2023-08-09:
All of the joystick hardware interaction is handled via the driver station, which is closed source and we do not have control over it. The robot program only receives a standardized, limited set of data from the driver station.

Starlight220, 2023-09-17:
Can this be closed as a wontfix?

closed 2023-09-17 by calcmogul