wpilibsuite/allwpilib #5533: [Glass] Glass Crash with Specific Config

ohowe1, 2023-08-12:
**Describe the bug**
After placing the attached config into the global save location, using the Open File or Reset operation crashes glass. I can look into which part of the file is problematic when I have the chance

**To Reproduce**
Steps to reproduce the behavior:
1. Place attached config in the global save location
2. Launch Glass
3. Click `Workspace/Open File`, then select a folder or click `Workspace/Reset`

**Expected behavior**
It not to crash

**Desktop (please complete the following information):**
 - OS: macOS

[glass.json.txt](https://github.com/wpilibsuite/allwpilib/files/12326250/glass.json.txt) (using txt extension so GitHub allows it)

rzblue, 2023-08-12:
I've reproduced this issue on Windows 11 as well.

ohowe1, 2023-08-12:
The problematic part is an array, any array will crash glass. For example:
```json
{
  "AnArray": [
    0
  ]
}
```

PR #5535 (2023-08-12): [glass] Fix array crash when clearing existing workspace
Fixes #5533.
closed 2023-08-12 by PeterJohnson