wpilibsuite/allwpilib #5537: wpilibjIntegrationTests differences

KangarooKoala, 2023-08-14:
**Describe the question you have.**
Why does wpilibjIntegrationTests use a different testing structure? Should it be made to match the other java tests?

**Describe the reason for your confusion.**
I would expect that it would be easier to use the same testing pattern, especially since I think this is causing the tests in wpilibjIntegrationTests to not be run. (The CI only does `./gradlew build`, which runs the `test` tasks but not the `run` tasks)

**Additional context**
Summary of the differences:
| wpilibjIntegrationTests | Other subprojects |
| --- | --- |
| Uses hamcrest Matchers | Doesn't use hamcrest Matchers |
| Uses different junit implementations | Uses junit-jupiter |
| Tests run via `./gradlew run` | Tests run via `./gradlew test` |

Here are the specific junit-related dependencies the different projects use:
```gradle
// shared/java/javacommon.gradle
testImplementation 'org.junit.jupiter:junit-jupiter-api:5.8.2'
testImplementation 'org.junit.jupiter:junit-jupiter-params:5.8.2'
testRuntimeOnly 'org.junit.jupiter:junit-jupiter-engine:5.8.2'

// wpilibjIntegrationTests/build.gradle
implementation 'junit:junit:4.13.2'
implementation 'com.googlecode.junit-toolbox:junit-toolbox:2.4'
implementation 'org.apache.ant:ant-junit:1.10.12'
```

calcmogul, 2023-08-14:
I tried upgrading them to JUnit 5, but I got stuck on parameterized tests, and we planned on replacing them with the cross-connect tests in the long-term anyway. No movement has occurred since then.

https://github.com/calcmogul/allwpilib/tree/_ref-wpilibj-integration-test-junit-5

calcmogul, 2023-08-14:
To be honest, we don't even check the test results. We just make sure they compile.

closed 2023-08-23 by calcmogul

calcmogul, 2023-08-23:
Superseded by #5561.