wpilibsuite/allwpilib #5545: Error in SymbolExports.h on windows

autoantwort, 2023-08-16:
**Describe the bug**
See https://github.com/wpilibsuite/allwpilib/actions/runs/5884679471/job/15959845041?pr=5516 from https://github.com/wpilibsuite/allwpilib/pull/5516

**To Reproduce**
Steps to reproduce the behavior:
1. https://github.com/wpilibsuite/allwpilib/pull/5516

The error is:
```
[1/857] Generating generated/main/cpp/wpilib-128.png.cpp
[2/857] Building CXX object wpimath/CMakeFiles/wpimath.dir/src/main/native/cpp/geometry/Transform3d.cpp.obj
FAILED: wpimath/CMakeFiles/wpimath.dir/src/main/native/cpp/geometry/Transform3d.cpp.obj 
C:\ProgramData\chocolatey\bin\c++.exe -DWPILIB_EXPORTS -Dwpimath_EXPORTS -ID:/a/allwpilib/allwpilib/wpimath/src/main/native/include -ID:/a/allwpilib/allwpilib/wpiutil/src/main/native/thirdparty/fmtlib/include -ID:/a/allwpilib/allwpilib/wpiutil/src/main/native/include -ID:/a/allwpilib/allwpilib/wpiutil/src/main/native/thirdparty/memory/include -ID:/a/allwpilib/allwpilib/wpiutil/src/main/native/thirdparty/json/include -ID:/a/allwpilib/allwpilib/wpiutil/src/main/native/thirdparty/llvm/include -ID:/a/allwpilib/allwpilib/wpiutil/src/main/native/thirdparty/mpack/include -ID:/a/allwpilib/allwpilib/wpiutil/src/main/native/thirdparty/sigslot/include -isystem D:/a/allwpilib/allwpilib/wpimath/src/main/native/thirdparty/eigen/include -isystem D:/a/allwpilib/allwpilib/wpimath/src/main/native/thirdparty/gcem/include -O2 -g -DNDEBUG -Wall -pedantic -Wextra -Werror -Wno-unused-parameter -std=gnu++20 -MD -MT wpimath/CMakeFiles/wpimath.dir/src/main/native/cpp/geometry/Transform3d.cpp.obj -MF wpimath\CMakeFiles\wpimath.dir\src\main\native\cpp\geometry\Transform3d.cpp.obj.d -o wpimath/CMakeFiles/wpimath.dir/src/main/native/cpp/geometry/Transform3d.cpp.obj -c D:/a/allwpilib/allwpilib/wpimath/src/main/native/cpp/geometry/Transform3d.cpp
In file included from D:/a/allwpilib/allwpilib/wpimath/src/main/native/include/frc/geometry/Transform3d.h:7,
                 from D:/a/allwpilib/allwpilib/wpimath/src/main/native/cpp/geometry/Transform3d.cpp:5:
Error: D:/a/allwpilib/allwpilib/wpiutil/src/main/native/include/wpi/SymbolExports.h:189:33: error: expected primary-expression before '__attribute__'
  189 | EXPORT_TEMPLATE_TEST(MSVC_HACK, __declspec(dllexport));
      |                                 ^~~~~~~~~~
D:/a/allwpilib/allwpilib/wpiutil/src/main/native/include/wpi/SymbolExports.h:108:37: note: in definition of macro 'EXPORT_TEMPLATE_INVOKE_2'
  108 |   EXPORT_TEMPLATE_##which##_##style(export, )
      |                                     ^~~~~~
D:/a/allwpilib/allwpilib/wpiutil/src/main/native/include/wpi/SymbolExports.h:181:17: note: in expansion of macro 'EXPORT_TEMPLATE_INVOKE'
  181 |   static_assert(EXPORT_TEMPLATE_INVOKE(                                    \
      |                 ^~~~~~~~~~~~~~~~~~~~~~
D:/a/allwpilib/allwpilib/wpiutil/src/main/native/include/wpi/SymbolExports.h:189:1: note: in expansion of macro 'EXPORT_TEMPLATE_TEST'
  189 | EXPORT_TEMPLATE_TEST(MSVC_HACK, __declspec(dllexport));
      | ^~~~~~~~~~~~~~~~~~~~
Error: D:/a/allwpilib/allwpilib/wpiutil/src/main/native/include/wpi/SymbolExports.h:108:45: error: expected primary-expression before ')' token
  108 |   EXPORT_TEMPLATE_##which##_##style(export, )
      |                                             ^
D:/a/allwpilib/allwpilib/wpiutil/src/main/native/include/wpi/SymbolExports.h:106:3: note: in expansion of macro 'EXPORT_TEMPLATE_INVOKE_2'
  106 |   EXPORT_TEMPLATE_INVOKE_2(which, style, export)
      |   ^~~~~~~~~~~~~~~~~~~~~~~~
D:/a/allwpilib/allwpilib/wpiutil/src/main/native/include/wpi/SymbolExports.h:181:17: note: in expansion of macro 'EXPORT_TEMPLATE_INVOKE'
  181 |   static_assert(EXPORT_TEMPLATE_INVOKE(                                    \
      |                 ^~~~~~~~~~~~~~~~~~~~~~
D:/a/allwpilib/allwpilib/wpiutil/src/main/native/include/wpi/SymbolExports.h:189:1: note: in expansion of macro 'EXPORT_TEMPLATE_TEST'
  189 | EXPORT_TEMPLATE_TEST(MSVC_HACK, __declspec(dllexport));
      | ^~~~~~~~~~~~~~~~~~~~
Error: D:/a/allwpilib/allwpilib/wpiutil/src/main/native/include/wpi/SymbolExports.h:108:3: error: 'EXPORT_TEMPLATE_TEST_MSVC_HACK_DEFAULT' was not declared in this scope; did you mean 'EXPORT_TEMPLATE_DECLARE_DEFAULT'?
  108 |   EXPORT_TEMPLATE_##which##_##style(export, )
      |   ^~~~~~~~~~~~~~~~
D:/a/allwpilib/allwpilib/wpiutil/src/main/native/include/wpi/SymbolExports.h:106:3: note: in expansion of macro 'EXPORT_TEMPLATE_INVOKE_2'
  106 |   EXPORT_TEMPLATE_INVOKE_2(which, style, export)
      |   ^~~~~~~~~~~~~~~~~~~~~~~~
D:/a/allwpilib/allwpilib/wpiutil/src/main/native/include/wpi/SymbolExports.h:181:17: note: in expansion of macro 'EXPORT_TEMPLATE_INVOKE'
  181 |   static_assert(EXPORT_TEMPLATE_INVOKE(                                    \
      |                 ^~~~~~~~~~~~~~~~~~~~~~
D:/a/allwpilib/allwpilib/wpiutil/src/main/native/include/wpi/SymbolExports.h:189:1: note: in expansion of macro 'EXPORT_TEMPLATE_TEST'
  189 | EXPORT_TEMPLATE_TEST(MSVC_HACK, __declspec(dllexport));
      | ^~~~~~~~~~~~~~~~~~~~
[3/857] Building CXX object imgui/CMakeFiles/imgui.dir/__/_deps/implot-src/implot.cpp.obj
ninja: build stopped: subcommand failed.
Error: Process completed with exit code 1.
```

calcmogul, 2023-08-16:
Probably another compiler flag issue, because this comiles fine with Gradle.

autoantwort, 2023-08-16:
I accidentally used gcc on windows :)

closed 2023-08-16 by autoantwort