wpilibsuite/allwpilib #5547: Replace J*ArrayRef with JSpan<T, Size>

calcmogul, 2023-08-16:
**Is your feature request related to a problem? Please describe.**
J*ArrayRef only supports const refs to data. This works fine for input parameters, but doesn't work for output parameters. The ArrayElements call in the destructor uses JNI_ABORT so any changes aren't copied back to Java.

**Describe the solution you'd like**
Replace J*ArrayRef with a templated JSpan class.

If the template type is non-const, JSpan would use JNI_ABORT in the destructor to discard native writes. If the template type is const, JSpan would use JNI_COMMIT in the destructor.

Create class specializations for `jdouble`, `jdoubleArray`, etc.

Use `if constexpr (std::is_const_v<T>)` to select the JNI_ flag in the destructor, and add is_const_v requires clauses to non-const accessor functions.

PeterJohnson, 2023-08-17:
Should also add a `data()` member function for consistency with the standard library.  It's worth noting that a non-const JSpan may be slower than SetXArrayRegion() for output parameters, particularly for larger arrays, as it may result in an extra allocation and copy being made of the array contents from Java to C++.

KangarooKoala, 2023-08-19:
Looking at the [Java 11 JNI docs](https://docs.oracle.com/en/java/javase/11/docs/specs/jni/functions.html), using `JNI_COMMIT` in the `Release<PrimitiveType>ArrayElements` function will "copy back the content but not free the `elems` buffer". Would a mode of 0 make more sense? (Docs say it will "copy back the content and free the `elems` buffer")

PeterJohnson, 2023-08-19:
Whoops yep, 0 would be correct.

PR #5554 (2023-08-24): [jni] Add JSpan and CriticalJSpan
Fixes #5547 

Notes:
* This probably fixes `DriverStationJNI.getAllJoystickData` and `DMAJNI.readDMA`, which wrote to an array which was then released with `JNI_ABORT`.
* Previously, `JByteArrayRef` did not have an `operator std::span<jbyte>()`. Was this intentional?
* The issue said to have `is_const_v` requires clauses for non-const accessors, so I included those, but I'm not sure they're necessary (since if the type is const-qualified, the return type is the same as the const accessor).
* Should `errs()` be piped `"\n"` or `std::endl`?
* This currently does [... cut]
closed 2023-08-24 by PeterJohnson