wpilibsuite/allwpilib #5548: Add a non repeatable schedule option for default commands.

TapChap, 2023-08-17:
During the last season, my team and I wanted to have sort of a reset command for one of our robot's subsystems.
after each use of the subsystem, we will call the `resetCommand` once, so the subsystem is ready for the next usage, but when we tried to figure out how this could be done, we realized we could add `.andThen(resetCommand)` to each of the subsystems main commands, or optionally just set the `resetCommand` as the subsystem's default command, which will then obviously cause it to be scheduled after any of the subsystem's commands finish, although it would be scheduled continuously every time it finishes, which wasn't ideal for the structure of the command...

what we ended up doing was something along the lines of this:
`subsystem.setDefaultCommand(subsystem.resetComand.andThen(new RunCommand()-> {}, subsystem);`
which did in fact give us the intended result but wasn't really a convenient solution..

What I'm suggesting is to add an overloaded method to the setDefaultCommand method in the Subsystem Interface that has another boolean parameter that allows to choose whether the command should be scheduled repeatedly or just once when the subsystem "frees" from any other command

Starlight220, 2023-08-17:
> what we ended up doing was something along the lines of this:
subsystem.setDefaultCommand(subsystem.resetComand.andThen(new RunCommand()-> {}, subsystem);
which did in fact give us the intended result but wasn't really a convenient solution..

In past discussions about possible refactoring of default commands, this was the sensible solution. What isn't convenient about it? Would factories help?

The solution you offer involves changing scheduler internals and adding a lot of complexity that is easily solved by the above; what benefit does it provide over this alternative?

TapChap, 2023-08-17:
Honestly, I suggested this feature because what we did felt like a bit of a workaround and not the "correct" way to do it, and I wondered if maybe there could be a better way to do it.
Although this fix was quite apparent to our team it might be more of a challenge for newer teams to figure out, so I thought about adding a more "official" way to get the same functionality

Starlight220, 2023-08-18:
Perhaps an `idle()` factory for the `RunCommand` part would make it more intuitive?
```java
command.andThen(Commands.idle())
```

TapChap, 2023-08-19:
That does seems like it should do the trick..