wpilibsuite/allwpilib #5558: Add examples/tests for command-based use cases

Starlight220, 2023-08-21:
Command-based is rather lacking in unit tests, and even more in integration tests. We should add tests and/or tested examples for use cases, as many and as varied as possible. This would help two problems at once: it would provide example code for various use cases, and ensure we don't break use cases (such as what happened with the proxying-defer-inside-composition case).
This is an invitation to comment here with any use case that you can think of. If you also have code that implements that use case -- even better.

KeithRieck, 2023-10-28:
When creating new example projects within VS Code,  the projects do not include a `src/test/java` directory, and no example unit tests are added to the project.  Specifically, the example unit tests in [wpilibjExamples](https://github.com/wpilibsuite/allwpilib/tree/main/wpilibjExamples/src/test/java/edu/wpi/first/wpilibj/examples) ought to be included, but don't appear.

Starlight220, 2023-10-29:
That's been fixed by #5838 and will be included in the next release.

ThadHouse, 2023-10-29:
> That's been fixed by #5838 and will be included in the next release.

It’s only fixed on the allwpilib side. It still needs to be fixed on the vscode side.