wpilibsuite/allwpilib #5604: Optional CommandGroup requirement union & scheduler changes

AngleSideAngle, 2023-09-03:
Currently, CommandGroups exhibit undesirable behavior when used in conjunction with critical default commands.

For example, if an arm subsystem requires a default command to stay raised and is used in a complex auto, the default command will not be triggered.

To solve this, there should be an option to avoid union requirements in sequential and parallel command groups.

As for implementation, the most obvious one would be modifying the command scheduler to update `m_requirements` from every scheduled command periodically, instead of only on init. That would allow command groups to simply forward their currently running command's requirements to the scheduler. It would also improve deferred commands in #5566 by allowing implicit runtime resolution of requirements instead of passing them through the constructor.

A potential alternative would be having the setting proxy every command in the group.

PeterJohnson, 2023-09-03:
I think a more robust way to fix this is to not use default commands to handle this case. The approach is to instead run the outputs only from periodic, and use commands only to communicate the desired state to the periodic code.  This avoids this issue entirely as long as you're disciplined never to run the outputs directly from any command, as it's unnecessary to have a default command running to hold the position (the periodic code handles this instead).

Starlight220, 2023-09-03:
Mutable requirements (or any other command property) would be a catastrophe.

For the use case you're describing, possible solutions are what @PeterJohnson suggested, explicitly including the default holding command in the composition, having the command hold position until interrupted, and I'm sure there are other ideas I'm not thinking of at the moment.

AngleSideAngle, 2023-09-03:
What specific issues would mutable command requirements cause?