wpilibsuite/allwpilib #5669: Implement more accurate pose estimation using factor graphs

calcmogul, 2023-09-19:
**Is your feature request related to a problem? Please describe.**
We currently use standard deviations of odometry and vision poses in a linear Kalman filter, but that's a poor fit for the true probability distribution. For example, the measurement noise in terms of distance increases with range to the tag.

**Describe the solution you'd like**
Factor graphs using [GTSAM](https://github.com/borglab/gtsam/). It can be compiled for the roboRIO like this:
```bash
cmake -B build -S . --toolchain=~/wpilib/2023/roborio/toolchain-config.cmake -DGTSAM_ENABLE_BOOST_SERIALIZATION=OFF -DGTSAM_USE_BOOST_FEATURES=OFF
```

JoshuaAN, 2023-09-19:
This is the current architecture for pose estimation

[image]

This is sub-optimal because feeding a normal distribution for pixels to solvepnp creates a nonlinear distribution for the output pose, while the kalman filter assumes a normal distribution.

The desired architecture is

[image]

The solvepnp math is done implicitly in the factor graphs solver, but the cost function is on pixels instead of output pose;
combined with odometry measurements at the same time.

calcmogul, 2023-09-19:
This also helps reduce pose ambiguity with AprilTags since the odometry is weighted a lot more than pixels in that case (since pixel variance is small for a wide range of poses).

calcmogul, 2023-12-21:
Declan mentioned recently we could try making the measurements the corner pixels with the measurement model in a UKF mapping from pose to corner pixels (so camera model, basically).