wpilibsuite/allwpilib #5678: Add frcYear to new commands, Romi, xrp vendordeps

sciencewhiz, 2023-09-22:


Starlight220, 2023-09-23:
What is the motivation for this? Historically it's been beneficial that the vendordeps haven't needed to be modified from year to year.

ThadHouse, 2023-09-23:
It’s extremely rare that vendor deps don’t have to be updated between years. And the breaks most do have are so hard to catch that it’s worth it to just require updating all.

srimanachanta, 2023-10-09:
For the vendordeps json, the wpilib version should also be included due to mid-season patches that might cause one vendor dep to use a different version of wpilib than the user code.

ThadHouse, 2023-10-09:
We're keeping it simple for now. Going that deep requires a lot of code changes, and is easy to screw up. Ideally we shouldn't be breaking mid year anyway. So this is only to solve the between year cases.

PR #5747 (2023-10-10): [commands,romi,xrp] Add frcYear to vendordep
Required by wpilibsuite/native-utils#170.
Also see wpilibsuite/native-utils#168.
Fixes #5678.
closed 2023-10-10 by PeterJohnson