wpilibsuite/allwpilib #5690: Fix intermittent crash on example unit test exit in CI

calcmogul, 2023-09-29:
The following C++ unit test executable fails in Windows CMake CI occasionally. The tests themselves pass, so it seems to be something that happens upon exit.
```
       Start 20: UnitTest
20/20 Test #20: UnitTest .........................***Exception: SegFault  0.03 sec
HAL Extensions: No extensions found
[==========] Running 4 tests from 1 test suite.
[----------] Global test environment set-up.
[----------] 4 tests from IntakeTest
[ RUN      ] IntakeTest.DoesntWorkWhenClosed
[       OK ] IntakeTest.DoesntWorkWhenClosed (0 ms)
[ RUN      ] IntakeTest.WorksWhenOpen
[       OK ] IntakeTest.WorksWhenOpen (0 ms)
[ RUN      ] IntakeTest.Retract
[       OK ] IntakeTest.Retract (0 ms)
[ RUN      ] IntakeTest.Deploy
[       OK ] IntakeTest.Deploy (0 ms)
[----------] 4 tests from IntakeTest (0 ms total)

[----------] Global test environment tear-down
[==========] 4 tests from 1 test suite ran. (0 ms total)
[  PASSED  ] 4 tests.

95% tests passed, 1 tests failed out of 20

Total Test time (real) =  14.16 sec

The following tests FAILED:
	 20 - UnitTest (SEGFAULT)
Errors while running CTest
```

Starlight220, 2023-09-30:
Does it also fail in gradle?

Gold856, 2026-11-13:
https://github.com/wpilibsuite/allwpilib/actions/runs/43501194325/job/132147362678#step:10:3394
The test failed on Bazel in an interesting way:
```
TIMEOUT: //wpilibcExamples/src/test/cpp/examples/UnitTest:UnitTest (Summary)
      D:/_bazel/execroot/_main/bazel-out/x64_windows-dbg-ST-e2d8886aad98/testlogs/wpilibcExamples/src/test/cpp/examples/UnitTest/UnitTest/test.log
INFO: From Testing //wpilibcExamples/src/test/cpp/examples/UnitTest:UnitTest:
==================== Test output for //wpilibcExamples/src/test/cpp/examples/UnitTest:UnitTest:
HAL Extensions: No extensions found
Warning: Bazel shard configuration is missing 'TEST_SHARD_INDEX'. Shard configuration is skipped.
Warning: Bazel shard configuration is missing 'TEST_TOTAL_SHARDS'. Shard configuration is skipped.
Warning: Bazel shard configuration is missing 'TEST_SHARD_STATUS_FILE'. Shard configuration is skipped.
Randomness seeded to: 2606719470
===============================================================================
All tests passed (4 assertions in 4 test cases)

Assertion failed: this->isReferenceToStorage(CI) && "Iterator to erase is out of bounds.", file D:\_bazel\execroot\_main\wpiutil\src\main\native\thirdparty\llvm\include\wpi/util/SmallVector.hpp, line 779
================================================================================
```

It timed out (60s timeout) as well.