wpilibsuite/allwpilib #5710: Add a way to trim each test directly via start and end time dual-pole slider

calcmogul, 2023-10-01:
If something like an arm hits a hard-stop, it throws off the fit. There's not an obvious way to trim that out, so teams are editing the analysis JSON manually. We should make that easier by letting the user directly control the start and end time for each of the four test runs.

We can compute defaults for them using the same algorithms we do now.

* Quasistatic test
  * Start time: motion threshold
  * End time: full test duration
* Dynamic test
  * Start time: motion threshold or time at which max acceleration is reached, whichever is later
  * End time: Time at which steady-state velocity is reached or full test duration, whichever is sooner

KangarooKoala, 2024-03-08:
How does this relate to the change in SysId? (Specifically, should this be configured in some way through the code instead of the now non-existent GUI?)

calcmogul, 2024-03-08:
Wdym non-existent GUI? The SysId tool still exists.

KangarooKoala, 2024-03-08:
Whoops, I guess I wasn't paying close enough attention to the SysId rewrite.