wpilibsuite/allwpilib #5717: Make wpimath state-space APIs support Eigen::Dynamic

calcmogul, 2023-10-02:
This would let RobotPy wrap them directly with Numpy arrays.

StateSpaceUtil already somewhat supports this. One of the main migration items there was to replace direct usage of template arguments for matrix row/column sizes with calls to the constexpr rows() and cols() functions.

LucienMorey, 2025-04-26:
Can I throw my hat in to help with this without cutting someone's grass? I would like to explore some state space stuff with my team, but the robotpy stuff is a touch brittle with specific specialisations at the moment.

calcmogul, 2025-04-27:
Feel free to pick any subissue to work on. https://github.com/wpilibsuite/allwpilib/issues/7257 is probably simplest to start.

LucienMorey, 2025-04-28:
I had a go at the suggested issue in #7942