wpilibsuite/allwpilib #5726: Joystick button presses not being logged

blu28, 2023-10-04:
**Describe the bug**
In the datalog output, button presses are being logged as:
     226.332608,DS:joystick1/buttons,<invalid>
rather than as specific button press indications.

**To Reproduce**
Steps to reproduce the behavior:
1. Start logging. 
2. Press a button.
3. Convert log with DataLogTool to CSV list format.

**Expected behavior**
The log would show what button was pressed.

**Desktop (please complete the following information):**
 - OS: Windows 10
 - 
 - Project Information: WPILib Information:
Project Version: 2023.4.2
VS Code Version: 1.83.0
WPILib Extension Version: 2023.4.3
C++ Extension Version: Not Installed
Java Extension Version: 1.18.0
Java Debug Extension Version: 0.51.0
Java Dependencies Extension Version 0.23.0
Java Version: 17
Java Location: C:\Users\Public\wpilib\2023\jdk
Vendor Libraries:
    CTRE-Phoenix (v5) (5.30.4+23.0.11)
    REVLib (2023.1.3)
    WPILib-New-Commands (1.0.0)

If you want to see the code: https://github.com/ToughTechs151/ChargedUp2023/tree/mil_august

**Additional context**
This may be related to #4467

rzblue, 2023-10-18:
I can't reproduce this in either wpilib 2023.4.3 (library and Data Log tool) or 2024.1.1-beta-1 (library and Data Log tool) in simulation with the following code in robotinit:
```
DataLogManager.start();
DriverStation.startDataLog(DataLogManager.getLog(), true);
```
I also can't reproduce it with your code running in simulation (wpilib 2023.4.2, Data Log tool 2023.4.3)

Just to confirm, the issue occurs when using a 2023 season version of Data Log Tool?
Could you upload a copy of a log file that has this behavior? (replace `wpilog` extension with `log` to bypass the file type restriction)

Starlight220, 2023-12-22:
OP hasn't responded, closing as can't repro.

closed 2023-12-22 by Starlight220