wpilibsuite/allwpilib #5769: Missing module `jinja2`

valentinegb, 2023-10-16:
**Describe the bug**
I am using CMake to build the repository. Using the following command:

```shell
cmake -S . -B build -DCMAKE_INSTALL_PREFIX=/opt/homebrew/Cellar/wpilib/2023.4.3 -DCMAKE_INSTALL_LIBDIR=lib -DCMAKE_BUILD_TYPE=Release -DCMAKE_FIND_FRAMEWORK=LAST -DCMAKE_VERBOSE_MAKEFILE=ON -Wno-dev -DBUILD_TESTING=OFF -DCMAKE_OSX_SYSROOT=/Library/Developer/CommandLineTools/SDKs/MacOSX14.sdk
```

I get this error:

```
Traceback (most recent call last):
  File "/tmp/wpilib-20231016-3394-ellcrb/allwpilib-2023.4.3/ntcore/generate_topics.py", line 4, in <module>
    from jinja2 import Environment, FileSystemLoader
ModuleNotFoundError: No module named 'jinja2'
CMake Error at ntcore/CMakeLists.txt:11 (message):
  python and python3 generate_topics.py failed
```

**To Reproduce**
Steps to reproduce the behavior:
1. Install `cmake` (v3.27.7)
2. Install `opencv` (v4.8.1)
3. Install `openjdk` (v21)
4. Run the CMake command shown above

**Expected behavior**
Expected to build all the libraries listed in `README-CMAKE.md`.

**Desktop**
 - OS: macOS 14.1 Beta (23B5067a)
 - Project Information: WPILib v2023.4.3, CMake v3.27.7, OpenCV v4.8.1, OpenJDK v21

**Additional context**
In case the context matters, I am creating a Homebrew formula for WPILib.

calcmogul, 2023-10-16:
You need to install that Python module.

PeterJohnson, 2023-10-16:
We probably need to update the readme.  See the cmake Mac CI workflow here: https://github.com/wpilibsuite/allwpilib/blob/main/.github/workflows/cmake.yml#L46

valentinegb, 2023-10-16:
> You need to install that Python module.

That should probably be stated in `README-CMAKE.md`.

valentinegb, 2023-10-16:
> We probably need to update the readme. See the cmake Mac CI workflow here: https://github.com/wpilibsuite/allwpilib/blob/main/.github/workflows/cmake.yml#L46

Oh that's super helpful for what I'm doing, thank you!

PR #5771 (2023-10-17): [docs] Add jinja2 to CMake prerequisites
Fixes #5769.
commit da70e4c2 (2023-10-17): [docs] Add jinja2 to CMake prerequisites (#5771)

Fixes #5769.
closed 2023-10-17 by PeterJohnson