wpilibsuite/allwpilib #5777: `make` fails with `make: *** [all] Error 2`

valentinegb, 2023-10-18:
**Describe the bug**
After using `cmake`, running `make` in `build-cmake` (the directory I configured for `cmake`), it fails with the following message:

```
make: *** [all] Error 2
```

... and no other sort of error message as far as I can tell. Here is the `make` log that was produced:

[05.make.log](https://github.com/wpilibsuite/allwpilib/files/12973124/05.make.log)

**To Reproduce**
Steps to reproduce the behavior:
1. ```shell
   cmake \
       -S . \
       -B build-cmake \
       -DCMAKE_INSTALL_PREFIX=/opt/homebrew/Cellar/wpilib/2023.4.3 \
       -DCMAKE_INSTALL_LIBDIR=lib \
       -DCMAKE_BUILD_TYPE=Release \
       -DCMAKE_FIND_FRAMEWORK=LAST \
       -DCMAKE_VERBOSE_MAKEFILE=ON \
       -Wno-dev \
       -DBUILD_TESTING=OFF 
       -DCMAKE_OSX_SYSROOT=/Library/Developer/CommandLineTools/SDKs/MacOSX14.sdk
   ```
2. `cd build-cmake`
3. `make`

**Expected behavior**
Expected all libraries in repository to build.

**Desktop**
 - OS: macOS 14.1 Beta (23B5067a)
 - Project Information: WPILib v2023.4.3, CMake v3.27.7, OpenCV v4.8.1, OpenJDK v21, Python v3.12.0, Jinja2 v3.1.2, MarkupSafe v2.1.3

**Additional context**
I am compiling the project using a Homebrew formula.

PeterJohnson, 2023-10-18:
Maybe try using ninja instead of make (-G Ninja)?

ThadHouse, 2023-10-18:
```
make[2]: *** No rule to make target `cscore/OPENCV_JAR_FILE-NOTFOUND', needed by `cscore/CMakeFiles/cscore_jar.dir/java_compiled_cscore_jar'.  Stop.
make[1]: *** [cscore/CMakeFiles/cscore_jar.dir/all] Error 2
```

It's not finding OpenCV

valentinegb, 2023-10-18:
> ```
> make[2]: *** No rule to make target `cscore/OPENCV_JAR_FILE-NOTFOUND', needed by `cscore/CMakeFiles/cscore_jar.dir/java_compiled_cscore_jar'.  Stop.
> make[1]: *** [cscore/CMakeFiles/cscore_jar.dir/all] Error 2
> ```
> 
> It's not finding OpenCV

Ah okay, jeez, would be nice if that bit of information wasn't in the middle of everything else lol
Since this is a Homebrew formula, OpenCV is listed as a dependency and is installed by Homebrew. I would think it would still have symbolic links in the directory that the library would be expected to be in, though.

valentinegb, 2023-10-18:
CMake seems to have no trouble finding it, since if I remove the dependency, CMake fails, but otherwise it makes it through until Make is run.

ThadHouse, 2023-10-18:
Its possible the homebrew install of OpenCV doesn't include Java. I'm not familiar enough with it to know.

CMake will not actually error if the dependency is missing. Its going to generate whatever it can. So its still CMake not finding it that is the error.

valentinegb, 2023-10-18:
Looking at the formula so far, it doesn't look like it depends on Java at all, so you may be right: https://github.com/Homebrew/homebrew-core/blob/HEAD/Formula/o/opencv.rb

valentinegb, 2023-10-18:
Yep, looks like it's explicitly disabled: https://github.com/Homebrew/homebrew-core/blob/18c5d616c62e63f48e2aa52491bea1709ad72bfa/Formula/o/opencv.rb#L94

closed 2023-10-18 by valentinegb