wpilibsuite/allwpilib #5780: SwerveDrive Optimize returning angles over 90

Kevin-OConnor, 2023-10-18:
In a Command-Based swerve drive project SwerveModuleState.optimize is returning angles over 90 degrees in simulation on occasion (semi-frequently). This happens for input angles both above and below 90. No threading, AdvantageKit, etc. involved. Repo: https://github.com/FRCTeam238/SwerveBase

This presents as a Heisenbug. In instances where I could get the incorrect angle to persist, placing a breakpoint and stepping through the optimize function "flips" the angle back and behaves correctly. Many instances were transient and with no change to joystick inputs the angle would flicker back and forth between correct and incorrect, or only briefly display incorrect.
[image]

I can't see anything in a code inspection of SwerveModuleState.optimize that would allow this to happen.

KangarooKoala, 2023-10-18:
Could you please also print out `state.speedMetersPerSecond` (or just print out `state`- The `toString()` includes both the speed and angle) and `encoderRotation`? That should provide enough data to reproduce the issue.

Kevin-OConnor, 2023-10-18:
That would have been smart to look at the actual encoder value instead of assuming it was 0 because it should be. Looks like the bug is in Rev's sim library somewhere, occasionally returning very erroneous encoder values (e.g. 7e22) when no sim is occurring.

closed 2023-10-18 by Kevin-OConnor