wpilibsuite/allwpilib #5803: Make fails when trying to compile protobuf files

Gold856, 2023-10-22:
**Describe the bug**
When attempting to build with Make, Make will fail when it attempts to compile the protobuf files.

**To Reproduce**
Steps to reproduce the behavior:
1. Use CMake to generate build files in a new directory.
2. Run Make on wpimath.
3. Observe a `No such file or directory` error for a protobuf directory and multiple errors for `.pb.h` files.

**Expected behavior**
Make should build it.

**Desktop (please complete the following information):**
 - OS: Ubuntu 22.04 on WSL2

**Additional context**
Ninja works perfectly fine, so it's something specific to Make. I'm using Ninja for now, but it'd still like to have Make as a build option.

calcmogul, 2023-10-22:
We all use Ninja, so I doubt this will ever get solved.

Gold856, 2023-10-22:
I figured. Should I leave this issue open if someone wants to try solving it or should I close it?

calcmogul, 2023-10-23:
We should probably just document that Ninja is required in README-CMAKE.md, and why Make doesn't work.

PR #5867 (2023-11-04): Fix protobuf generation when building with make
For some reason, cmake does not create `PROTOC_OUT_DIR` when building using Make, so protoc can't generate code. This works around that by creating the appropriate folder.
fixes #5803
closed 2023-11-04 by PeterJohnson