wpilibsuite/allwpilib #5808: Upgrade JavaDocs to JDK 17

calcmogul, 2023-10-24:
There's a lot of warnings from `./gradlew docs:generateJavaDocs -PdocWarningsAsErrors` to fix, which are usually from missing docs.

You can usually find the docs for JNI a layer down, like in the HAL.

rzblue, 2023-10-24:
Specifically, every java symbol needs to be documented. This can (and should) be done over multiple PRs.

m10653, 2023-12-07:
I noticed that many of these Check Style warnings are due to Enums missing javadoc comments.  It does not seem like anywhere in wpilib the values of the enums themselves have comments and they can be a bit clunky. Is adding javadoc comments to the values of the enums desired or would modifying the style rules/annotations to ignore this style/docs violation preferred. 

### Example of these comments
```java
  public enum CalibrationTime {
    /** Calibration time of 32 milliseconds */
    _32ms(0), 
    /** Calibration time of 64 milliseconds */
    _64ms(1)
    }
    
```

Gold856, 2023-12-18:
Since [JDK 18](https://bugs.openjdk.org/browse/JDK-8275351), doclint can be supressed with the usual `SuppressWarnings`, which isn't helpful for us because we're on JDK 17. And updating to JDK 21 (the next LTS release) brings _even more_ warnings. So maybe ignore the enum stuff for now?

Gold856, 2023-12-31:
For posterity, #6119 removed about 3000 warnings.

Gold856, 2024-01-01:
#6132 removes 84 warnings.

PR #6220 (2024-01-20): [docs] Switch to JavaDoc 17 and add missing docs
Fixes #5808.
closed 2024-01-20 by PeterJohnson