wpilibsuite/allwpilib #5814: roboRIO Team Number Setter Tool Feature Requests

jonathandao0, 2023-10-24:
- The NI roboRIO Imaging Tool reboots the roboRIO automatically after changing the team number using the Edit Startup Settings option. The WPILib roboRIO Team Number Setter Tool should follow the same behavior or at least warn the user that the roboRIO should be rebooted after setting the team number.
- The roboRIO's serial number and comment should be added as a column in the displayed information table (See #4462).
  - If possible, it would be nice to be able to update the comment using the tool as well.
- Add the Reset button to the main window. Currently, it is hidden in the Workspace drop down menu, which feels a bit inconvenient. This would be useful for refreshing the roboRIO's information after it has been rebooted/updated.

ThadHouse, 2023-11-11:
The roboRIO actually doesn't need to reboot to take in the team number. Thats why it doesn't automatically reboot.

I can add the serial number, but probably won't add the comments. Those are easiest to update through the webpage, and dynamic formatting in imgui is hard.

Unsure what you mean by Reset. The reset button in the menu just resets the layout, and doesn't actually do anything related to the roboRIO connections. It should automatically be refreshing, but some OS's are a bit sticky and will hold onto old information, even after losing connection.

calcmogul, 2025-09-09:
OBE by 2027 no longer using the roboRIO.