wpilibsuite/allwpilib #5817: Dashboards do not re-enumerate values after disconnection and reconnection

rzblue, 2023-10-24:
**Describe the bug**
Dashboards (so far I've tested glass, shuffleboard, and outlineviewer) don't enumerate values after they are disconnected and reconnected (ie. by removing and replacing the rio ethernet cable)

Disabling and re-enabling the NT client connection in glass/outlineviewer appears to resolve the issue each time it happens.

**To Reproduce**
 - publish data to NT from roborio
 - Connect to roborio NT server using glass
 - observe entries in NT view
 - unplug rio ethernet cable
 - observe values disappear
 - plug rio ethernet cable back in
 - values do not come back (but log and title bar indicate a reconnection)

**Expected behavior**
Values appear back in the NT view after reconnection

**Additional context**
The rio I've tested with is currently not on an frc control system; It's on my home network with a 192.168.1.X IP addresss. However, i've tested connection discovery with both mdns and giving glass the IP directly with no behavior change, so i don't believe this has anything to do with the issue.

Starlight220, 2023-10-24:
Maybe this is somewhat causing the Shuffleboard values disappearing issue?

PeterJohnson, 2023-10-24:
I believe this issue is new this year.  It only affects Windows so it probably related to the networking changes, but I need to debug it.

closed 2023-10-29 by PeterJohnson

AngleSideAngle, 2024-01-17:
My team continues to experience this in WPILib 2024.1.1 on a Windows 10 laptop.

PR #5841 (2023-10-29): [wpinet] WebSocket: Fix write behavior
On Windows, TryWrite will always return 0 if there is a Write in progress. The previous behavior for SendFrames and SendControl just used a normal Write, which caused issues with code that combined these with TrySendFrames. Instead, have SendFrames and SendControl also use TryWrite under the hood if possible, and create write requests if not. The implementation preserves the priority of SendControl against an existing write request with multiple frames.

Fixes #5817 (the root cause for that was it was not sending the subscription frame due to the WS ping  [... cut]