wpilibsuite/allwpilib #5825: [ci] Tracking issue: Update actions to Node 20

rzblue, 2023-10-25:
GitHub has added deprecation warnings for actions that run with Node 16. A lot of the actions we use haven't been updated upstream; This issue is so we don't wait to the last minute to update them.

- [ ]  mozilla-actions/sccache-action (cmake, sanitizers)
- [ ] actions/checkout (all of them) (#5736)
- [ ] actions/setup-python (cmake, lint-format, upstream-utils, comment-command)
- [ ] actions/github-script (comment-command)
- [ ] actions/setup-java (lint-format, gradle, documentation, comment-command)
- [ ] lukka/get-cmake@v3.27.6 (cmake, sanitizers)
- [ ] lukka/run-vcpkg@v11.1 (cmake, sanitizers)
- [ ] actions/upload-artifact@v3 (gradle, lint-format)
- [ ] webfactory/ssh-agent@v0.8.0 (documentation)
- [ ] wpilibsuite/import-signing-certificate@v1
- [ ] gradle/wrapper-validation-action (gradle/wrapper-validation)

sciencewhiz, 2023-12-04:
Given the growing pains that were associated with the Node 16 transition, and the fact that it looks like we'll be able to get through the season without updating, the WPILib teams doesn't plan to attack this until later, but PRs are certainly welcome.

spacey-sooty, 2023-12-05:
:+1: I'll try open prs up for everything so when the wpilib team wants to transfer they can just merge everything in

spacey-sooty, 2024-05-04:
I believe github script, setup python, setup java and gradle wrappper validation action are all done on allwpilib and have prs open for other repos

Gold856, 2024-06-20:
#6325 covered actions/setup-python, actions/setup-java, and most of actions/github-script. #6692 covered lukka/get-cmake, lukka/run-vcpkg, and most of mozilla-actions/sccache-action. #6758 covers the rest.

PR #6758 (2024-06-22): [ci] Bump actions versions
Bump webfactory/ssh-agent to 0.9.0 (Node 20)
Switch to gradle/actions/wrapper-validation (Node 20)
Bump mozilla-actions/sccache-action to 0.0.5 (Node 20 for sanitizers only)
Bump actions/github-script to 7 (Node 20 for documentation only)
Bump wpilibsuite/import-signing-certificate to 2 (Node 20)
Bump JamesIves/github-pages-deploy-action to 4.6.1 (Node 20)
Resolves #5825.
closed 2024-06-22 by PeterJohnson