wpilibsuite/allwpilib #5853: [cscore] "Manual exposure property write failed" and friends

mcm001, 2023-11-01:
**Describe the bug**

With beta-2 and a laptop webcam (that previously worked), I get the following exceptions when setting manual exposure:

```
[2023-10-31 22:41:21] [VisionModule - VisionSourceSettables] [ERROR] Failed to set camera exposure!: property write failed
[2023-10-31 22:41:21] [VisionModule - VisionSourceSettables] [ERROR] VideoException [edu.wpi.first.cscore.VideoException: property write failed]
        at edu.wpi.first.cscore.CameraServerJNI.setCameraExposureManual(Native Method)
        at edu.wpi.first.cscore.VideoCamera.setExposureManual(VideoCamera.java:74)
        at org.photonvision.vision.camera.USBCameraSource$USBCameraSettables.setExposure(USBCameraSource.java:193)
        at java.base/jdk.internal.reflect.NativeMethodAccessorImpl.invoke0(Native Method)
        at java.base/jdk.internal.reflect.NativeMethodAccessorImpl.invoke(NativeMethodAccessorImpl.java:77)
        at java.base/jdk.internal.reflect.DelegatingMethodAccessorImpl.invoke(DelegatingMethodAccessorImpl.java:43)
        at java.base/java.lang.reflect.Method.invoke(Method.java:568)
        at org.photonvision.vision.processes.VisionModuleChangeSubscriber.onDataChangeEvent(VisionModuleChangeSubscriber.java:186)
        at org.photonvision.common.dataflow.DataChangeService.dispatchFromQueue(DataChangeService.java:65)
        at java.base/java.lang.Thread.run(Thread.java:833)
```

V4l2-ctrls:
```
matt@photonvision ~/D/G/photonvision (2023_10_15-gstreamer) [1]> v4l2-ctl --list-ctrls -d 2

User Controls

                     brightness 0x00980900 (int)    : min=0 max=255 step=1 default=128 value=209
                       contrast 0x00980901 (int)    : min=0 max=255 step=1 default=32 value=32
                     saturation 0x00980902 (int)    : min=0 max=100 step=1 default=64 value=64
                            hue 0x00980903 (int)    : min=-180 max=180 step=1 default=0 value=0
        white_balance_automatic 0x0098090c (bool)   : default=1 value=0
                          gamma 0x00980910 (int)    : min=90 max=150 step=1 default=120 value=120
                           gain 0x00980913 (int)    : min=4 max=8 step=1 default=4 value=4
           power_line_frequency 0x00980918 (menu)   : min=0 max=2 default=2 value=2 (60 Hz)
      white_balance_temperature 0x0098091a (int)    : min=2800 max=6500 step=1 default=4000 value=4000
                      sharpness 0x0098091b (int)    : min=0 max=7 step=1 default=2 value=2
         backlight_compensation 0x0098091c (int)    : min=0 max=2 step=1 default=0 value=0

Camera Controls

                  auto_exposure 0x009a0901 (menu)   : min=0 max=3 default=3 value=1 (Manual Mode)
         exposure_time_absolute 0x009a0902 (int)    : min=10 max=2500 step=1 default=156 value=2500
     exposure_dynamic_framerate 0x009a0903 (bool)   : default=0 value=1
```

kernel: `Linux photonvision 6.2.0-36-generic #37~22.04.1-Ubuntu SMP PREEMPT_DYNAMIC Mon Oct  9 15:34:04 UTC 2 x86_64 x86_64 x86_64 GNU/Linux`

Using photonvision 2024.1.1-beta-1

**Desktop (please complete the following information):**
 - OS: Ubuntu 22.04

Starlight220, 2023-11-01:
Caused by #5478?

PeterJohnson, 2023-11-01:
Closing as duplicate.

closed 2023-11-01 by PeterJohnson