wpilibsuite/allwpilib #5882: OpenCV unsatisfied link error in windows simulation

mjansen4857, 2023-11-05:
**Describe the bug**
Unable to use any of the OpenCV library in windows simulation after the beta 3 update. Using opencv will result in an unsatisfied link error:
```
Error at org.opencv.core.Mat.n_Mat(Native Method): Unhandled exception: java.lang.UnsatisfiedLinkError: 'long org.opencv.core.Mat.n_Mat()'
        at org.opencv.core.Mat.n_Mat(Native Method)
        at org.opencv.core.Mat.<init>(Mat.java:23)
        at frc.robot.Robot.robotInit(Robot.java:31)
        at edu.wpi.first.wpilibj.TimedRobot.startCompetition(TimedRobot.java:106)
        at edu.wpi.first.wpilibj.RobotBase.runRobot(RobotBase.java:363)
        at edu.wpi.first.wpilibj.RobotBase.lambda$startRobot$0(RobotBase.java:436)
        at java.base/java.lang.Thread.run(Thread.java:833)
```

I've tried clearing my gradle caches and that seems to have no effect.

**To Reproduce**
Steps to reproduce the behavior:
1. Create a new timed robot template project
2. Add this line to robot init: `Mat test = new Mat();`
3. Run in simulation (on windows, not sure about other platforms), program crashes with above error

**Desktop (please complete the following information):**
 - OS: Windows 11
 - Project Information:
  ```
  WPILib Information:
Project Version: 2024.1.1-beta-3
VS Code Version: 1.84.0
WPILib Extension Version: 2024.1.1-beta-3
C++ Extension Version: 1.17.5
Java Extension Version: 1.23.0
Java Debug Extension Version: 0.52.0
Java Dependencies Extension Version 0.23.0
Java Version: 17
Java Location: C:\Users\Public\wpilib\2024\jdk
Vendor Libraries:
    WPILib-New-Commands (1.0.0)

  ```

ThadHouse, 2023-11-05:
This actually isn't a new issue, and has existed for years. The issue is that OpenCV doesn't by default load its JNI library. We have some tricks to make it do so, but they're not done correctly.

You can work around this for now by creating a CvSink or CvSource before using raw OpenCV objects.

```
CvSink sink = new CvSink("Name");
```

It doesn't even need to be used, it just needs to be constructed.

ThadHouse, 2023-11-05:
Actually, see https://www.chiefdelphi.com/t/unsatisfied-link-error-when-simulating-java-robot-code-using-opencv/426731/3 I had forgotten we had come to that decision. So yeah you just have to use one of the cscore cv objects first.

mcm001, 2023-11-05:
Skull.png but makes sense, easy enough to fix on our end

closed 2023-11-05 by mjansen4857