wpilibsuite/allwpilib #5884: Call to resetOdometry() in getAutonomousCommand() is not made when SwerveControllerCommand is executed as it should

gartaud, 2023-11-05:
**Describe the bug**
The call to resetOdometry() in getAutonomousCommand() is not made when SwerveControllerCommand is executed as it should. Rather it is made when getAutonomousCommand() is called.

This is not good as resetOdometry() requires the current pose prior to the execution of the SwerveControllerCommand.

For more info, see discussion at
https://www.chiefdelphi.com/t/how-to-define-the-shortest-trajectory-in-reverse-with-edu-wpi-first-math-trajectory-trajectorygenerator/443549/20

rzblue, 2023-11-05:
The lines in question:
https://github.com/wpilibsuite/allwpilib/blob/main/wpilibjExamples/src/main/java/edu/wpi/first/wpilibj/examples/swervecontrollercommand/RobotContainer.java#L114-L118

We should also check other path following examples to make sure this isn't the case in those as well.

ElliotScher, 2023-11-09:
I'm a little confused as to where the odometry is to be reset? Is the odometry supposed to be reset in the initialize method of SwerveControllerCommand? Sorry, I'm new to contributing to wpilib.

gartaud, 2023-11-09:
Hello

In response to “I'm a little confused as to where the odometry is to be reset? Is the odometry supposed to be reset in the initialize method of SwerveControllerCommand? Sorry, I'm new to contributing to wpilib.”

It would probably be better NOT to change SwerveControllerCommand for teams that want to use absolute coordinates in auton.

 

Rather the examples should be updated so that another command is called to reset the odometry prior to SwerveControllerCommand being called (either via an explicit command, or via a lambda)

 

Thanks.

ElliotScher, 2023-11-10:
@gartaud Could this be achieved with a `.beforeStarted()` decorator?

gartaud, 2023-11-10:
Seems like a good idea.